hd-zero/hdzero-goggle #1: Better spectrum analyzer

Remleyfpv, 2022-08-18:
A better spectrum analyzer in the menu that allows you to see the real time RF noise. Even though there already is one it isnt very nice bc you can only see HDZero channels. It would be nice to be able see the RF usage of all things on the 5.8 spectrum.

TwistedFPV, 2023-01-06:
This would be a very useful feature all around. Especially for whoops, we have this nice whoop lite bundle for guys that want hd whoops. Whoops can sometimes be a pain to find outside. Larger quads can have a decent motor beeper but whoops not so much, this would be ideal. Its a feature i always loved having on my rapidfire. If you ever lost a aircraft, you know how much it sucks. Flying in new places, you can get disoriented and think you crashed in one area but really your not even close. Its one of those features youll be glad to have. Besides that, knowing what's noise is in thw area, so you can avoid it and be on a good channel right off the bat.

TwistedFPV, 2023-02-26:
Bump
We need this!

pitts-mo, 2023-08-02:
I would like the ability to see general RF usage across the band (spectrum analyzer view), show the HDZero channels at their approximate placements within the band, show if HDZero signal is detected in those channels, and still allow selecting from HDZero channels that do not already show an HDZero signal was detected.

MrMap, 2023-10-23:
It´s a lovely idea to have a more sophisticated spectrum analyzer in the goggles. I am mostly into FPV Combat (like Laser tag on RC planes). Often I get the role of VTX channel manager.

In this setting I use my goggles in an spectral environment of multi user, multi system brands, sometimes legacy stuff, and it can be really challenging and confusing to manage a workable and dynamic channel allocation, even when consulting available tables for systems, bands and channels. Manufacturers at some degree have their own proprietary nomenclature of bands and channels, partly due to regulatory restrictions - inviting for allocation fuckups. And some goggle systems emit telemetry (in the pilot area). Sometimes I make a stupid mistake myself, after telling all the guys to be careful with the channel settings...

Having an intelligent analyzer in the goggles would be GOLD and IMHO another selling point.