hd-zero/hdzero-goggle #11: Source of the media.tar

razzeee, 2022-12-18:
What's the source/license of the media.tar? Why is it distributed like this?

bkleiner, 2022-12-18:
they are libraries for different media functions directly from allwinner ([semi-official source](https://github.com/lindenis-org/lindenis-v536-softwinner/tree/master/eyesee-mpp))
i presume they are distributed like this because the build process in the original source's build process is incredibly convoluted.

i have started an attempt to sort that out [here](https://github.com/bkleiner/hdzero-goggle/tree/toolchain/lib/softwinner) some binary blobs remain however.

razzeee, 2022-12-19:
Hrm, do you know which allwinner chip ended up in there? I only want to see if av1 is possible, but if it's not supported in hardware it's probably a no go. (I don't even know if they figured that out for their recent chips)

bkleiner, 2022-12-19:
it's a allwinner v536, AV1 might be a little ambitious tho, in terms of video-codec support lifecycles that is basically brand new.

razzeee, 2022-12-19:
I know, still the target with hardware support for av1 was 2021/22 if I remember right from talking with the devs on release in 2018. But I would have expected allwinner to not be on the leading edge according to my previous knowledge.

razzeee, 2022-12-19:
Thanks!

Summary:
- av1 is impossible, as the chip is from 2019
- @bkleiner is working on a rework of the build process

closed 2022-12-19 by razzeee