hd-zero/hdzero-goggle #12: Why are we using h264?

razzeee, 2022-12-20:
Hey, 

I'm wondering, why we're using  h264 and not h265, from a quick reading, it seems like hevc should be capable of lower data rates and we should have hardware support on the googles.

https://github.com/hd-zero/hdzero-goggle/blob/main/src/player/vdec2vo.c#L30

Cheers

bkleiner, 2022-12-29:
As far as i can tell the line you referencing is merely a suggestion to the **decoder** and the files produced by the DVR are indeed encoded in hevc and also correctly decoded by aforementioned decoder. 
[image]

razzeee, 2022-12-29:
Hrm, I'm more interested in the vtx to Google Transmission and what's used for that

bkleiner, 2022-12-29:
none! hdz does not use any compression for transmission. instead a custom asic was designed for low-latency transmission of the image directly on a baseband chip. This is all handled pretty much entirely handled in hardware, and very much out of the scope of this repository. refer to the [divimath website](http://www.divimath.com/products/) and [hdzero website](https://www.hd-zero.com/document) for a little more information.

razzeee, 2022-12-29:
Thanks, will do that. Kinda hoped there would be some things to tinker with :)

closed 2022-12-29 by bkleiner