hd-zero/hdzero-goggle #19: Goggle MP4 file not recognized by QuickTime or FinalCutPro

sunolfpv, 2022-12-29:
Goggle DVR can record in MP4 format. However, file is not recognized as valid by Apple/Mac video editing software. Mac users still need to use Handbrake, ffmpeg or VLC to access the files, similiar to a TS file.
The advantage of the MP4 format was to be able to directly import into video editing software, but this is not the case for Mac users.

bkleiner, 2022-12-29:
Let’s keep this open for future reference, but to me it really sounds like you should file an issue with Apple. Sounds to me QuickTime (probably what finial cut uses under the hood) is not reading mp4 files, other programs are completely fine with.

sunolfpv, 2022-12-29:
Apple programs can read MP4. Apple does some weird stuff, but the mp4/h264 is based on quicktime and Apple are license holders/contributors to h264/h265. You can use Apple programs like Compressor to output h265 in MP4.  I don't know what the delta is between these files and something native from Apple, but first time I have seen this issue.

bkleiner, 2022-12-29:
I did _not_ say apple programs can not read mp4 _at all_. It seems QuickTime refuses however to read files that are considered valid by other programs _even on the same platform_, which makes it QuickTimes issue in my book.

sunolfpv, 2022-12-29:
Alf on the HDZero Discord provided this work around. Posting it here in case other Mac users see this and need a fix. You can use VLC or Handbrake to transcode, but this method with ffmpeg is instant and lossless. I'm not a developer, but I think it is just changing the metadata. Only potential gotcha is you need to use the command line.

After installing ffmpeg, the command to use is 'ffmpeg -i input.mp4 -vcodec copy -acodec copy -tag:v hvc1 output.mp4' where input.mp4 is your original file and output.mp4 is whatever you want to name it.

Here is a video on installing ffmpeg from scratch: https://www.youtube.com/watch?v=nmrjRqEIgGc

Here is a sample of it being run:

 ffm % ffmpeg -i hdz_004.mp4 -vcodec copy -acodec copy -tag:v hvc1 hdz_ffm.mp4
ffmpeg version 5.1.2 Copyright (c) 2000-2022 the FFmpeg developers
  built with Apple clang version 14.0.0 (clang-1400.0.29.202)
  configuration: --prefix=/opt/homebrew/Cellar/ffmpeg/5.1.2_1 --enable-shared --enable-pthreads --enable-version3 --cc=clang --host-cflags= --host-ldflags= --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libbluray --enable-libdav1d --enable-libmp3lame --enable-libopus --enable-librav1e --enable-librist --enable-librubberband --enable-libsnappy --enable-libsrt --enable-libtesseract --enable-libtheora --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libspeex --enable-libsoxr --enable-libzmq --enable-libzimg --disable-libjack --disable-indev=jack --enable-videotoolbox --enable-neon
  libavutil      57. 28.100 / 57. 28.100
  libavcodec     59. 37.100 / 59. 37.100
  libavformat    59. 27.100 / 59. 27.100
  libavdevice    59.  7.100 / 59.  7.100
  libavfilter     8. 44.100 /  8. 44.100
  libswscale      6.  7.100 /  6.  7.100
  libswresample   4.  7.100 /  4.  7.100
  libpostproc    56.  6.100 / 56.  6.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'hdz_004.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2mp41
    encoder         : Lavf59.16.100
  Duration: 00:02:42.72, start: 0.000000, bitrate: 22178 kb/s
  Stream #0:0[0x1](und): Video: hevc (Main) (hev1 / 0x31766568), yuvj420p(pc, bt470bg/bt470bg/smpte170m), 1280x720, 22175 kb/s, 60 fps, 60 tbr, 15360 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
Output #0, mp4, to 'hdz_ffm.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2mp41
    encoder         : Lavf59.27.100
  Stream #0:0(und): Video: hevc (Main) (hvc1 / 0x31637668), yuvj420p(pc, bt470bg/bt470bg/smpte170m), 1280x720, q=2-31, 22175 kb/s, 60 fps, 60 tbr, 15360 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
Stream mapping:
  Stream #0:0 -> #0:0 (copy)
Press [q] to stop, [?] for help
frame=    1 fps=0.0 q=-1.0 size=       0kB time=00:00:00.00 bitrate=5415.4kbits/frame= 9763 fps=0.0 q=-1.0 Lsize=  440522kB time=00:02:42.70 bitrate=22180.4kbits/s speed= 738x    
video:440474kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.010710%

Radla, 2023-01-05:
Instead of ffmpeg you can use the tool avidemux, which also offers basic editing on keyframes without reencoding.
Simple select copy for both, audio and video and select the prefered containerformat.

bkleiner, 2023-01-07:
in #62 the tag QuickTime seems to take issue with was added to the record app.
this change is also included in the latest full release [here](https://www.hd-zero.com/document)
can you check if the problem persists?

sunolfpv, 2023-01-07:
Fixed! Awesome - works great - seamless now in QuickTime or FinalCutPro.

closed 2023-01-07 by sunolfpv