hd-zero/hdzero-goggle #23: Goggle does not Auto Scan  on boot up (firmware 2212022)

MartinHugh, 2022-12-31:
Existing Behaviour
=============
With Auto Scan setting set to
    Auto Scan = On
    Default      = HDZero

When powering up the goggles, the user is taken to the Scan menu.
In order to scan for a transmitting video signal from an aircraft, the user is forced to trigger a scan by pressing the left centre button.

Expected Behaviour
=============
When powering up the goggles with Auto Scan = On, the goggles should Auto Scan and give the user a choice of channels which were detected on the scan.

If Auto Scan = Off, and Default is set to HDZero, the goggles should automatically attempt to tune to the last HDZero channel which was selected on a previous session.

bkleiner, 2023-01-01:
This is an interesting issue, because its not a bug, but an intentional change based on user feedback.
Initially the goggles would start scanning if AutoScan = On, however it was perceived as annoying that the goggles did not allow any menu navigation until the scan was complete. 
So the behaviour was changed, to not start the scan unless the key was pressed. I think its important we do not re-introduce the behaviour that was perceived as annoying by others. 

With @pkendall64 current proposal there is a small time window where you can navigate away before the scan starts (caused by other changes???), but it does indeed still lock you into the scan screen as soon as it has started.
i think we need to figure out if that is good enough or if we need to look into a different solution.

MartinHugh, 2023-01-01:
> This is an interesting issue, because its not a bug, but an intentional change based on user feedback. Initially the goggles would start scanning if AutoScan = On, however it was perceived as [...]

My interpretation of the proposal implemented is that the user can now configure in a third state of autoscan, which allows the menu to be displayed without scanning.   Presumably, this would satisfy the requirement you have described (showing the scan menu without scanning) as well as allowing the goggles to be set for actual autoscan on bootup.

bkleiner, 2023-01-01:
yup, brought this up with some of the beta-testers, indeed seems the new 3rd option should cover all bases

PR #25 (2023-01-02): Change autoscan to on/last/off
Changes the "Auto Scan" option to `On`/`Last`/`Off`

- `On` does the auto scan
- `Last` tunes to last channel
- `Off` just displays the main menu on the scan page

Fixes #23
closed 2023-01-02 by bkleiner