hd-zero/hdzero-goggle #26: Feature Request: DVR Record HDMI Source

shannonbaker, 2023-01-01:
When using a HDMI source it would be helpful to record this input.  This is useful for keeping DVR recordings in a single spot when used with complementary VRXs such as the WS VRX.

Suggest minimal goggle OSD whist the recording happens (record, goggle temp warning and goggle battery warning) with goggle OSD not recorded.

bkleiner, 2023-01-03:
we gonna need some input from divimath on this one.

CrazyDoge128, 2024-01-07:
That would be a great feature. I was thinking of it when i got my ws vrx.

robhaswell, 2024-01-31:
I'm convinced that my goggles used to record HDMI DVR on a previous firmware, but they don't at the moment.

ligenxxxx, 2024-06-06:
Due to copyright issues, we will not add this feature.

closed 2024-06-06 by ligenxxxx