hd-zero/hdzero-goggle #27: Feature Request: Double tap menu button to switch to and from expansion module (analogue) input.

shannonbaker, 2023-01-02:
Double tap the menu button to take the goggle to expansion module input without needing to enter menus and select.  This is ideal for pilots whom fly long range and retain analogue video as well as digital on a craft so they can switch between easily during flight.

shannonbaker, 2023-01-06:
Perhaps have a menu option to alter the jog wheel function from switching channels to switching input sources.

uPz3TT3r, 2023-01-16:
+vote for this! 
Alternative: option in menu to change input with function button instead of changing fan speed.

bkleiner, 2023-01-18:
Lets keep the interactions well defined. Double tap cycles through all import sources.

a-di-ez, 2023-01-31:
@bkleiner I think it's better to double-click on the OK button of the wheel to call a pop-up window with a list of all sources and select the desired one with the wheel, then after a pause of 2 seconds automatically or by pressing OK - select this channel.

This is how we get rid of:
- the need to scan only connected sources
- from accidentally switching the source instead of entering the menu
- accidentally pressing twice, which will cause the source to switch, just wait a couple of seconds or press the OK button on the wheel and the selection menu disappears

And: 
- it's faster to select only what you need than to jump over all sources

Similarly, it would be convenient to adjust the fan speed: we hold the button for a couple of seconds and the menu for changing the fan speed from 1 to 10 appears, since now the most minimal fan mode creates a lot of noise and a couple of points below are missing to make the glasses silent. Plus it will speed up the fan speed reset after we blew the lenses, instead of switching all the speeds in a circle.

What do you think?

bkleiner, 2023-01-31:
> @bkleiner I think it's better to double-click on the OK button of the wheel to call a pop-up window with a list of all sources and select the desired one with the wheel, then after a pause of 2 [...]

similar to https://github.com/hd-zero/hdzero-goggle/issues/98 and yeah i agree that some flavor "context menus" are probably what we ultimately want, however there is some "plumbing work" that needs doing so we can make that happen.

a-di-ez, 2023-01-31:
@bkleiner yes, exactly that
I understand that implementing a popup on top of a video is not as easy as just going to the main menu
I'm already starting to think about learning another programming language, namely C, to help you, as such products are very inspiring

ligenxxxx, 2024-06-06:
Switching sources is a dangerous thing to do and is best done from the menu.

closed 2024-06-06 by ligenxxxx