hd-zero/hdzero-goggle #31: Optional goggle power off at battery voltage

sunolfpv, 2023-01-03:
Optional menu setting under Power that turns goggle completely off at a given voltage or after no input signal detected for a given time. Not just a warning, but completely off. Idea is to save the screens from burn in or the battery past 3.0v, for example, if goggle is accidentally left on and unattended.

bkleiner, 2023-01-06:
The oleds already automatically turn off when the goggles do not detect any more movement for a given time.
 
As far as i can tell, we have no means of _completely_ powering off the goggles, when i shutdown the linux running on the goggles the fans etc. are still running, so that wouldn't be all that helpful for not killing the battery.

sunolfpv, 2023-01-06:
Thank you for the background. Closing as not relevant given OLED shut off and not being able to physically turn off from OS.

closed 2023-01-06 by sunolfpv