hd-zero/hdzero-goggle #36: Store VTX Firmware Locally to make flashing easier

KaminKevCrew, 2023-01-04:
It would be great if we could store the file that contains the firmware for each VTX locally on the goggles in their zipped format, and select which VTX we're flashing in the goggle menu then have the goggle unzip that file, flash the VTX, and delete the HDZERO_TX.bin file from the SD card, so I can flash the next VTX.

bkleiner, 2023-01-04:
feature is planned, however i am not gonna bother with zips, rough plan is to allow suffixes (eg `_WHOOP_LITE.bin`)

KaminKevCrew, 2023-01-04:
Ah, that would be great! I am definitely looking forward to it. 

Another idea that came up in the Discord is that once the Wifi module is available, it could potentially just download whatever firmware the user wants, and can then flash the VTX after downloading. I'm not sure if that will be possible, but if it is that would be a really neat feature.

aviphysics, 2023-01-19:
Everytime I have to update, I wish this was a thing. I think I have one of every VTX, so it is always a total pain.

closed 2024-06-06 by ligenxxxx