hd-zero/hdzero-goggle #37: Focus chart screen to aid IPD/focus adjustment

MBoerlage, 2023-01-04:
Request to add an option to display a full screen "**Back Focusing Chart**" image (black & white) to help with manual IPD/Focus adjustment. 

1. Select "Focus Chart" in goggle menu (to be added)
2. Display full screen image of "Back Focusing Chart" - its a black and white chart wheel in the center and small ones in each corner
3. Allow user to adjust mechanical IPD/Focus nobs on goggle
4. Use a button to return to the goggle menu, return to menu

The image should be stored in firmware (not on SD card), several "Back focus chart" examples can be found on goggle. Since its sole purpose is focus, it is preferred over colored TV test screens. It is crucial the image has small circles on each corner and one in the middle. 

Example applications are ORQA Pilot.One goggles, they have this as an option "Deep focus" which is awesome. Some users adjust IPD/focus looking at the camera footage (inaccurate) or their OSD on their camera footage (depending on where your OSD is, inaccurate). A back focusing chart will make sure you get the best sharpness out of your setup.

riveraj082, 2023-01-04:
This is a good idea.

MBoerlage, 2023-01-05:
[image]

EastCoaster, 2023-01-05:
Excellent idea, maybe even include the steps needed to dial in focus and IPD.

MBoerlage, 2023-01-06:
> i have never heard of these before, but does indeed seem like a useful addition. do you think you could provide a 1920x1080 version of this chart? i assume we would want it to fill the whole screen.

Please find attached - with many thanks and credits to Ian from the Houston FPV race community, he made one from scratch. Feel free to suggest any changes/adds. 

[image]

bkleiner, 2023-01-06:
> Please find attached - with many thanks and credits to Ian from the Houston FPV race community, he made one from scratch. Feel free to suggest any changes/adds.
>
> [image]

Awesome, thank you!

bkleiner, 2023-01-06:
i have create a initial implementation for this, you can find binaries to try here https://github.com/hd-zero/hdzero-goggle/actions/runs/3857514237
be advised that this feature builds on some experimental changes, so you want to revert back to a stable release for continued use.

limonspb, 2023-01-06:
> i have create a initial implementation for this, you can find binaries to try here https://github.com/hd-zero/hdzero-goggle/actions/runs/3857514237 be advised that this feature builds on some experimental changes, so you want to revert back to a stable release for continued use.

nice!
Is this firmware based on 122.8?

bkleiner, 2023-01-06:
> > i have create a initial implementation for this, you can find binaries to try here https://github.com/hd-zero/hdzero-goggle/actions/runs/3857514237 be advised that this feature builds on some experimental changes, so you want to revert back to a stable release for continued use.
> 
> nice! Is this firmware based on 122.8 + experimental features?

jup, 7.68.122.8 is fully merged.

limonspb, 2023-01-06:
> i have create a initial implementation for this, you can find binaries to try here https://github.com/hd-zero/hdzero-goggle/actions/runs/3857514237 be advised that this feature builds on some experimental changes, so you want to revert back to a stable release for continued use.

seems to be working :)
[image]
[image]

MBoerlage, 2023-01-06:
yall are awesome!

bkleiner, 2023-01-06:
_personally_ i find the white background pretty harsh on the eyes (those oleds are just so bright!)
should we invert the colors? have the background be black and the sections be white? 
would that reduce the versatility in any way?

MBoerlage, 2023-01-06:
> _personally_ i find the white background pretty harsh on the eyes (those oleds are just so bright!) should we invert the colors? have the background be black and the sections be white? would that reduce the versatility in any way?
> 
> like this ![dark]

Yeah that's a good idea - be careful with the conversion (above was a bit smudged in the center) or let us know if you need help with that.

wetheredge, 2023-01-06:
Looks like you've got a good inverted copy, but here's an SVG I made if anyone wants to further customize colors:

[image]

Just need to edit the source in a text editor, then it should be able to be rasterized by GIMP or similar.

bkleiner, 2023-01-07:
build of the latest version with the inverted screen here https://github.com/hd-zero/hdzero-goggle/actions/runs/3859396827

commit bab04dc5 (2023-01-07): add focus chart page fixes #37
closed 2023-01-07 by bkleiner