hd-zero/hdzero-goggle #41: Feature request: When no signal is detected, give the user control of what to show instead of the 'rainbow' pixels

AlexandreBonneau, 2023-01-05:
Add an option so that user can select which type of 'display' is shown when 'no signal' is detected.

ie. when no signal is detected, currently it first shows a screen full of 'rainbow pixels'.
It would be nice to be able to select if the user wants to see either a full color that you could choose (ie. black, dark blue, etc.), or a picture that the user could upload beforehand (ie. a race logo).

bkleiner, 2023-01-05:
We gonna have to look into whether or not this is viable. AFAIK the rainbow screen occurs when the ASIC has _not yet_ communicated that the link was lost.

bkleiner, 2023-02-19:
Regrettably not an option - it's made like this in the silicon.

closed 2023-02-19 by bkleiner