hd-zero/hdzero-goggle #42: Feature request: Improve the DVR filename convention

AlexandreBonneau, 2023-01-05:
Please change the video files naming convention for something like `2023-01-24 16h46 HDZ_0042.ts` (in the VRX as well), so that the files could be ordered chronologically (alphabetically), even if they come from multiple HDZero goggle (which would then not have the same index counter).

Also, make sure the extension is `.ts`, not the uppercase `.TS`.

bkleiner, 2023-01-05:
This repository is about the goggle. The Goggle does not have a functioning RTC, prepending a timestamp is not an option. Is there another proposal?

AlexandreBonneau, 2023-01-05:
A correct timestamp is the only thing I think would greatly improve DVR files organization.
Perhaps others can chip in.

dadog92, 2023-01-06:
what about getting the RTC from the radio from the ERLS backpack?

AlexandreBonneau, 2023-01-09:
According to Ryan Quellet in is [disassembling video](https://youtu.be/p4qXiQrtCkU?t=450), the goggle do support RTC, but do not have a powersource.
Assuming some users will be willing to open up their goggle to add one, then it makes senses to rename the DVR file with the timestamp info, if the RTC in powered up.

bkleiner, 2023-01-09:
> According to Ryan Quellet in is [disassembling video](https://youtu.be/p4qXiQrtCkU?t=450), the goggle do support RTC, but do not have a powersource. Assuming some users will be willing to open up their goggle to add one, then it makes senses to rename the DVR files with the timestamp info mentioned in the first comment, if the RTC is powered up.

looking forward to your pull request implementing this feature without negatively impacting users that did not perform this mod and does not generate heaps of support tickets (why does the RTC not work when i enable it in the settings?)

AlexandreBonneau, 2023-01-09:
No need to post snarky comments really.
The point of feature requests, are they are...requests.
My C skills are quite dusty, though I can help with making sure the UX is on point.

About the settings, why even create one; if you detect a powered on RTC, use its info, it not, do not.

bkleiner, 2023-01-09:
right - you are submitting issues to a open source repository for features to be implemented by people like me.
These are basically crude technical documents, a technical discussion, where the point is to come up with an plan to implement a given feature.
This discussion, unless you know something i dont and you are planning on implementing it yourself, does not seem to lead there - you seem to treat it more like a Wishlist and that is not the point imho. As you say yourself, these are _requests_. 
[This recent request](https://github.com/hd-zero/hdzero-goggle/issues/37) is a good example on i feel these should go. 

Yes, in a perfect world, we would simply be able to detect whether the RTC battery is present or not, i have not gotten around to checking if that is a function we have available. That seems like a good solution to that problem - and it might have been obvious to you - but that would be part of what needs writing down in these.

Wikiwix, 2023-01-24:
@bkleiner regarding your idea for the RTC check.
I would personally not check for the battery being available, but for the DateTime being larger than a threshold:

* I suspect the clock will reset to a specific point in time if power is lost
* I suspect that point in time is years or at least months in the past
* That means there is no danger of suspecting a correct time without someone setting the time on purpose (powering the goggles for multiple months seems unlikely)
* If someone wants a timestamp without an RTC battery, that would be feasible (if he sets the clock after every restart)

I will update #116 (hopefully this evening) with a more concrete implementation suggestion taking my suggestion into account.

bkleiner, 2023-01-24:
> @bkleiner regarding your idea for the RTC check. I would personally not check for the battery being available, but for the DateTime being larger than a threshold:
>
> * I suspect the clock will reset [...]

I really don't know if there is a point to even enabling the functionality without the battery present. Feels to me would mostly just lead to a ton of complaints about the goggles not keeping time from people not realizing they need a battery or forgetting to update the time after boot. That's just bad UX in my mind.
Otherwise that proposal seems sound to me. Threshold could just be date of the Goggle announcement.

Wikiwix, 2023-01-26:
@bkleiner I updated the request in #116 and created #123 as requested.

I put my suggestion on how this can look from a UX perspective there.

I hope that description tackles your reservations you have regarding users without a RTC battery.

PR #379 (2024-01-05): Add date-based naming scheme for DVR files and add video metadata
This PR allows users with an active RTC to chose between the conventional naming scheme (i.e. hdz_000, hdz_001 etc.) and a date-based one (i.e. 2023-11-24T13_34_37).
When no rtc battery is present, the user can not change the naming scheme. On top of that, it automatically falls back to the conventional scheme as default.

Further, I added some metadata to the DVR files i.e. a title (currently the file name) and - for mp4 - the date it was created.
Over the course of my research and the implementation, I found out that the ts co [... cut]
closed 2024-01-05 by ligenxxxx