hd-zero/hdzero-goggle #44: Feature request: Add a PiP mode (Picture-in-picture)

AlexandreBonneau, 2023-01-05:
Add a PiP (Picture in picture) mode to display sources in a small rectangular window, on top of the main feed (ie. Main analog + pip HDZero, Main HDZero + pip analog, etc.).

This could be very interesting in Races if you use HDZero as the main feed, and get an additional Analog feed from the referee that would broadcast the times and position of everybody, or if racing in teams the feed from your teammate.

This could also be useful of bigger quad with a second analog rear camera.

Ideally should be able to change the size of such PiP display, and move it around the screen to best fit the user needs.

kevinsumner, 2023-04-04:
This would also be useful for situations like using a "cheater" quad (camera in the front and camera in the back) for lining up rear-facing chase shots.

schmufu, 2023-09-17:
I have attached an analog FPV camera and have it on the AV input. It would be nice to have a picture-in-picture of the camera even when flying to keep an eye on my environment.

ligenxxxx, 2024-06-06:
HDZero Goggle cannot support this.

closed 2024-06-06 by ligenxxxx

AlexandreBonneau, 2024-06-06:
> HDZero Goggle cannot support this.

Does this means the hardware cannot stream the HDZero feed *and* the external analog module at the same time?
Or just that this feature is not currently worked on/wanted by the HDZero team?