hd-zero/hdzero-goggle #48: Feature request: Stream the video to an external device, wirelessly

AlexandreBonneau, 2023-01-05:
I've seen/heard this feature request before, but could not find it mentioned here, so here goes:
Using the the WiFi module to stream the video seen on the goggle screen, to an external device would be awesome.

Perhaps create an rstp endpoint, so that smartphone or computers or the same WiFi network could access the live streaming?

This would allow the live stream to be processed and recorded directly in an external device, and could be used to add effects to the video stream live.

bkleiner, 2023-01-05:
personally i was thinking we go a little more modern with HLS and/or DASH, a raw MPEG-TS stream might be another option.

Master92, 2023-11-11:
Couldn't this issue be closed? Video streaming over the external module already works for some time now 😉

ligenxxxx, 2024-06-06:
Solved.