hd-zero/hdzero-goggle #50: Feature request: Display the time and date on the OSD

AlexandreBonneau, 2023-01-05:
Most of the quads out there do not have a RTC, hence there are no dates/times in the DVR.
It would be useful to have an option to put a small timestamp on the top of the OSD.
For instance, `2023-01-14 16:27`.

If this is to much for some pilots, perhaps only display that information on disarm (not sure the goggles can know when the quad is armed, perhaps via the VTX?).

bkleiner, 2023-01-05:
The goggles currently do not possess a functioning RTC because there is no backup battery (simplifies shipping).
We can revisit this if somebody finds a way to activate it.

closed 2023-01-05 by bkleiner

AlexandreBonneau, 2023-01-05:
What does "finds a way to activate it" means? The hardware is there, but needs a constant power source?
If that's the case, maybe somebody could point out where to solder what voltage input?

In the meantime, this feature request could be transferred to the VTX repository when it'll be created, since the VTX does have an RTC + battery.

bkleiner, 2023-01-05:
> What does "finds a way to activate it" means? The hardware is there, but needs a constant power source? If that's the case, maybe somebody could point out where to solder what voltage input?

as i understand it, the SoC in the goggles has a internal RTC, it is unknown however if those pins are exposed.
however i think if that changes the situation any, because most users wont crack open the goggles to solder on a battery, making this feature not viable for general availability.

> In the meantime, this feature request could be transferred to the VTX repository when it'll be created, since the VTX does have an RTC + battery.

what do you mean by 'when it'll be created', the VTX firmware has been opensource for a while. https://github.com/hd-zero/hdzero-vtx
to my knowledge, the second part is not correct, i don't think the VTX has a RTC and there is no permanent powersource.

AlexandreBonneau, 2023-01-05:
Sorry my bad, I meant to write VRX, not VTX.