hd-zero/hdzero-goggle #51: Selectable behavior for handling corrupt pixel information

kevinsumner, 2023-01-05:
Currently, corrupted pixels show up as a random color, leading to a rainbow static across the image. This has often been seen as an issue, particularly folks coming from other systems.

Assuming it is possible to detect corrupt pixels (I don't know if the HDZero protocol has any kind of checksum or other error-detection), it would be nice to be able to select from one of a few different error handling behaviors, aside from the current. Some possible options:

- Color error pixels black or a medium grey.
- If the information for the pixel from the previous frame was not corrupted, reuse that information. In the event of multiple, sequential errors for a pixel, fall back to another method.
- The nearest neighbor method discussed in #40.

AlexandreBonneau, 2023-01-05:
Selecting what to display could follow the same options mentioned in https://github.com/hd-zero/hdzero-goggle/issues/41

bkleiner, 2023-01-05:
This is really outside of the scope of _this_ repository, as the image is received and processed by a asic & fpga.
The SoC only composes the OSD and displays it.

closed 2023-01-05 by bkleiner