hd-zero/hdzero-goggle #52: Feature request: Use the headtracking feature for VR

AlexandreBonneau, 2023-01-05:
It would be great to be able to able to use the gyro headtracking information and feed it into a computer, so the goggle could be used like a VR headset.

I'm not sure exactly how such information could be sent (and with what latency), perhaps via the Wifi module?

Do note that this feature request, in my opinion, is very, very low priority.

kyleireland, 2024-06-27:
I believe HDZero has the same image per eye unlike a VR having two slightly offset images.

JSMiCh, 2025-06-15:
> I believe HDZero has the same image per eye unlike a VR having two slightly offset images.

What if it's possible to disable the displays individually? Then you could show frames alternately for each eye, which would give a total of 45 frames per second.