hd-zero/hdzero-goggle #54: Bug: Playback Icon Incorrect After Fast Forward/Rewind

sabshire, 2023-01-05:
When using the playback feature, if the video is played, then paused, and then fast forward/rewind is used, video begins to play, but the playback icon is incorrect.  User must tap the play/pause button twice in order to get the icon correct, and to then pause the video from playing.  If the intended function is to play upon fast forward/rewind, the icon should correctly indicate that.  Better functionality would be to allow for fast forward/rewind while paused and maintain the paused state.

bkleiner, 2023-01-05:
happend to be working on the ui code, if you would help this along you could help by testing https://github.com/hd-zero/hdzero-goggle/actions/runs/3848877809

sabshire, 2023-01-05:
How do I do that?

bkleiner, 2023-01-05:
download and extract the bin file from the artifacts section on the page linked above, place on the sdcard like any other FW update.

sabshire, 2023-01-05:
Working now!  Thanks!

bkleiner, 2023-01-05:
:+1: thank you! always good to have second set of eyes on something like that.
i will try to have these changes flow into a full release at the next juncture.
you might wanna consider flashing back to an official release, some aspects of that version are early days.

sabshire, 2023-01-05:
Will do.  Thanks again!

commit f0f47805 (2023-01-05): player: preserve playstate during seek, fixes #54
closed 2023-01-05 by bkleiner