hd-zero/hdzero-goggle #55: Feature request: Enable the Func Button to act as a return in the main menu

EastCoaster, 2023-01-05:
While navigating the menu, it would be nice to have the Func Button act as a back button. This saves time navigating around the menu eliminating the need to scroll to the bottom of a sub-menu to return to the previous menu. Still keep the current <Back in the menu as some people may prefer to scroll. Similar functionality is offered on the other goggles and would not disrupt the manual DVR recording.

bkleiner, 2023-01-05:
long pressing the primary button within the encoder already serves as a back button, which imho, is preferable because it allows for single hand operation.

EastCoaster, 2023-01-06:
I just checked the long hold and it works perfectly. Thank you for the feedback, I will close the ticket.

EastCoaster, 2023-01-06:
Long pressing the primary button returns the sub menu to the menu and enabling the func button to perform the same task is duplicative.

closed 2023-01-06 by EastCoaster