hd-zero/hdzero-goggle #57: Feature request: Support other filesystems for the SD card

AlexandreBonneau, 2023-01-05:
Right now, you need to use fat32, which is simple yes, but very limited and subject to fragmentation.

Since the goggles are using Linux, it would be great to allow reading and writing to sd card formatted with ext4, btrfs, f2fs, xfs, etc.

This would be transparent to Windows/Mac users, where they would just use their usual SD card.
For Linux users though, it would mean not having to use an obsolete FS, and fully use 256GB+ cards.

You could even support exfat for Windows users.

ligenxxxx, 2024-06-07:
This involves a system update and there are no plans to change it for now.

closed 2024-06-07 by ligenxxxx

AlexandreBonneau, 2024-06-08:
While there is no plan _now_, this feature request is still valid.
Instead of closing valid feature request which would enhance the product, could we not instead tag those with a "Todo later" tag instead?