hd-zero/hdzero-goggle #63: OSD kiss ultra not working correctly

nathanielakkermans, 2023-01-09:
Hi all, I don't know if this is the place for my issue but I give it a try. I have received my new hdzero goggles and that are very nice. But I have some problems with the fc osd. I use kiss ultra fc and it is working but on the new goggles the screen is full of lines (so a -) the osd elements are showing but there rest of the screen of full. On the old farshark Rx this is not the case and the osd is working fine. Updated the kiss fc but no difference, updated the vtx aan goggles but still the same. If I disable the osd in the fc then the lines are gone. Is this a known issue? 
Thanks in advance

bkleiner, 2023-01-09:
yes this is right place! could you try creating an image somehow?

nathanielakkermans, 2023-01-09:
Yes tonight I will add an image, I will also try a second quad with kiss and see if it has the same problem.

nathanielakkermans, 2023-01-09:
[image]
Here a picture, strange this is that my other quad don't have this issue. Both use the kiss ultra and both on the whoop vtx with runcam V2. The one from the picture I updated the vtx and it reports ver 1.0.0 in the vtx menu. The other quad reports v37. To be clear the quad with shows the problems I updated it because of this issue but it didn't solve it. But why it reports ver 1.0.0 and the other 37 is a bit strange I think.

Proskate, 2023-01-11:
Kiss uses Ardupilot font (or simmilar, for me ardu one works), so You need to upload custom HD font with name ARDU_000.bmp. Instructions are under hdzero-osd-font-library

bkleiner, 2023-01-12:
@nathanielakkermans can you verify if that is the case? fetch the ardu font here https://github.com/hd-zero/hdzero-osd-font-library and place it on the sdcard as described in the readme.

nathanielakkermans, 2023-01-21:
i'm sorry for the late replay. this did the trick! thank you for the solution. the strange thing is that one quad was working and the other had the strange osd lines. after the ardu font update the one with the lines was nice and fixt but i noticed that the other quad had a different font. so i checked the FC firmware and that one was running an old firmware 2.0.1 B23 (kiss ultra) and the other was running on 2.0.1 B43. so i guess that somewhere between these versions it changed witch osd it used. but all good now! thanks

bkleiner, 2023-01-21:
Neat! an updated ARDU font was also merged into this repo recently. that should fix the issue _without_ installing a custom font.

closed 2023-01-21 by bkleiner