hd-zero/hdzero-goggle #64: Devcontainer not working on Mac

creecedev, 2023-01-10:
When starting the remote dev container, the setup.sh runs and after the media is extracted, there's a CMake error that a simple test program cannot be compiled. The same process of cloning into the container and building on my Windows machine had no issues.

I am new to Docker in general but i'm doing the same thing on both systems and getting different results.

creecedev, 2023-01-10:
This appears to be related to M1 Macbooks. I found a stackoverflow that discusses the issue:

https://stackoverflow.com/questions/71040681/qemu-x86-64-could-not-open-lib64-ld-linux-x86-64-so-2-no-such-file-or-direc

The solution was to add --platform=linux/amd64 to the FROM line of the Dockerfile:

**FROM --platform=linux/amd64 mcr.microsoft.com/devcontainers/base:ubuntu**

I'm now able to successfully compile on my M1 Macbook. I'm not sure if this breaks anything if added for other systems but it may be worth adding if not.

bkleiner, 2023-01-10:
appears to me that linking itself succeeds, but something fails during packaging.
`qemu: uncaught target signal 11 (Segmentation fault) - core dumped` seems the place where it fails.
i dont see any usage of qemu in the build script. is that part of the M1 emulation somehow?

creecedev, 2023-01-10:
> 

Yes, it appears its something to do with M1. I'm just going to go back to Windows and develop there as I have no idea how to get this one resolved.

bkleiner, 2023-01-22:
turns out a lot of the problematic process was redundant. could you try https://github.com/hd-zero/hdzero-goggle/pull/111 and see if it now builds on M1.

Nikolas-S, 2023-02-10:
> turns out a lot of the problematic process was redundant. could you try #111 and see if it now builds on M1?

Not sure when it was fixed but works fine for me in an x86 container on M1, it's just really slow since it's emulated.
I tried getting an arm native toolchain to work and actually managed to get the build working in an arm64 virtualized container by using the arm gnu toolchain and kinda smashing it together with the x86 one. 
It builds a lot faster but I'd be surprised if it actually runs, if anyone feels like trying it, I take no responsibility. ^^
https://github.com/Nikolas-S/hdzero-goggle/tree/feature/arm64_host_build_support

bkleiner, 2023-02-10:
then let's close this one....toolchain "smashing" feels kinda dangerous to me.
if we were to implement a aarch64 native toolchain, we should probably just bake one ourselfs.

closed 2023-02-10 by bkleiner