hd-zero/hdzero-goggle #67: Bug: Inaccurate cell count

evilC, 2023-01-10:
Repro: 
1. In the power menu, set the voltage slider to 2.8v  
2. Power the goggles up from a 12v source (Simulating a 4S 18650 pack at 3v per cell)
3. The goggles show 3S
4. You will therefore not get a warning when the voltage drops to 11.2v (4 cells at 2.8v each)

Solution:
The cell count in the power menu needs to be able to be manually set

Nikolas-S, 2023-01-26:
I added a toggle and slider to the power settings page to switch between auto/manual mode and select the number of cells here: https://github.com/Nikolas-S/hdzero-goggle/tree/feature/manual_batt_s_count
I just preordered my goggles recently so if someone already has goggles and would be willing to test I would appreciate it, so I know if it works before creating a pull request.

bkleiner, 2023-01-26:
> I added a toggle and slider to the power settings page to switch between auto/manual mode and select the number of cells here: [...]

awesome, please open a PR right away, we will get it tested and fixed if need be, there.

closed 2023-01-27 by bkleiner

evilC, 2023-01-27:
Removed the bug aspect of this issue, will re-create in another issue so that can now be addressed

evilC, 2023-01-27:
Aww crap, I think I noticed a bug.
The `Cell Voltage` slider is now inoperable.
It's always in the "selected" state (bar is green), and when I select it, it exits the power menu

bkleiner, 2023-01-27:
> @Nikolas-S Aww crap, I think I noticed a bug. The `Cell Voltage` slider is now inoperable. It's always in the "selected" state (bar is green), and when I select it, it exits the power menu

are you sure you are running the latest version of that PR? voltage deal works fine here.

Nikolas-S, 2023-01-27:
> @Nikolas-S Aww crap, I think I noticed a bug. The `Cell Voltage` slider is now inoperable. It's always in the "selected" state (bar is green), and when I select it, it exits the power menu

Hmm that's interesting, taking a quick look at things I changed that might be relevant, I don't see what could cause that issue.
Is it repeatable?

evilC, 2023-01-27:
> are you sure you are running the latest version of that PR? voltage deal works fine here.

I was not aware that there was a newer version than the one I tested. 
I just re-flashed with [this one](https://github.com/hd-zero/hdzero-goggle/actions/runs/4016114228) and all is OK

bkleiner, 2023-01-27:
> > are you sure you are running the latest version of that PR? voltage deal works fine here.
> 
> I was not aware that there was a newer version than the one I tested. I just re-flashed with [this one](https://github.com/hd-zero/hdzero-goggle/actions/runs/4016114228) and all is OK

there wasn't. the ci builds every commit that is submitted via PR, you must have picked one from before my last round of review.

Nikolas-S, 2023-01-27:
I think what caused that issue was that I changed the max element count of the power setting page from 4 to 6 in the latest commit of the PR, if you were using a version where it was still 4, that would explain that behavior. @evilC