hd-zero/hdzero-goggle #68: Feature request: Option to show average cell voltage

evilC, 2023-01-10:
Please add an equivalent of BetaFlight's average cell voltage  
If you implemented #67 , then this would always be accurate-ish

ie, if (Once #67 is implemented) I have set the cell count set to 4S, and the current voltage of the pack is 12v, I want to see `3v` in the OSD for the voltage

uPz3TT3r, 2023-01-16:
vote for this! 
+Add Voltage in overlay menü of all inputs (HDZero, Analog). So that its possible to react early on a race and change battery during heats, not only if warning comes up. (in my opinion voltage is more important info than fan speed of top fan)

bkleiner, 2023-01-16:
i think both this and #67 are absolutely necessary features and should be implemented at the next juncture.
There is some "plumbing" work i would like to take care off first however.

evilC, 2023-01-16:
> vote for this! +Add Voltage in overlay menü of all inputs (HDZero, Analog). So that its possible to react early on a race and change battery during heats, not only if warning comes up. (in my opinion voltage is more important info than fan speed of top fan)

I agree

MartinHugh, 2023-02-07:
Just to be clear, if we are talking about displaying the "goggle" battery voltage, either total or cell - I dont mind, I think it is essential, and would very much welcome it.

closed 2023-02-20 by bkleiner

MartinHugh, 2023-02-20:
Is this supposed to show the voltage in the flight OSD, or in the menu.

I cant see it in the flight OSD.  Should I raise another issue for this ?

bkleiner, 2023-02-20:
PR supplies the battery logic, but does not modify _where_ it is displayed, as the _orginial_ feature request stated.
So if you would like to see it be displayed somewhere else please raise a new request for that.

MartinHugh, 2023-02-20:
> PR supplies the battery logic, but does not modify _where_ it is displayed, as the _orginial_ feature request stated. So if you would like to see it be displayed somewhere else please raise a new request for that.

Thanks

PR #162 (2023-02-20): Add option to display average cell voltage in osd
Added a toggle to the power settings page to choose whether the osd shows the total battery voltage or the average voltage per cell.
I wanted to make it obvious which mode is active, so "/C" is added to the old format when cell average mode is active, if someone has a better idea let me know.
e.g.:
"Total" mode:       4S 15.6V
"Cell Avg." mode: 4S 3.9V/C

Would appreciate if someone could test the pr on their goggles, don't have mine yet.

Resolves #68