hd-zero/hdzero-goggle #69: Feature request: warn user if googles are pointing to the ground

Combinacijus, 2023-01-10:
# Problem
When I fly usually my head little by little drops down, which mean patch antennas are pointing down and are useless

# Solution
Add setting for min and max google angle and if out of range for set amount of time warn user with visual or audio warning to correct his head

For example if patches have verticla fov of 30 deg you'd set angle_min: -15, angle_max: 15
If patches are pointed up 10deg then -25, 5 or whatever range user chooses

evilC, 2023-01-16:
I would definitely welcome this feature.
Especially if, when your head was pointing at the ground, and audio prompt plays (in a posh english voice) "Chin up, old chap!" 🤣

abacero, 2023-01-17:
I would second this. However, if possible, a live visual indicator on the OSD such as an angle +/- from 0 degrees would be preferred over an audio indicator. For instance, users have complained that Orqa's head tilt warning has a constant beeping when the goggles are hanging from one's neck or forehead.

SuperSniper1018, 2023-01-28:
> I would second this. However, if possible, a live visual indicator on the OSD such as an angle +/- from 0 degrees would be preferred over an audio indicator. For instance, users have complained that [...]

What if it somehow grabbed whether or not the quad was armed, via ELRS Backpack or something, and only gave the audio warning if armed?

abacero, 2023-02-03:
> What if it somehow grabbed whether or not the quad was armed, via ELRS Backpack or something, and only gave the audio warning if armed?

That would be a great idea. Not sure how feasible it would be for non-ELRS, but since the backpack is already built in, I would imagine it would be easier to implement. It would be cool if it can also be done via MSP, thus it would not be receiver protocol dependent.

Junnicutt, 2023-05-14:
There is a way to do this currently but it only works with the head tracker being setup and on. If you're using elrs and have the current master fw for goggles, goggle back pack, elrs tx, and elrs tx backpack then the head tracker can work. Once it's turned on the tx has your head tilt angle. From there it just takes a logical switch to check the chanel value of the tilt channel and a special function to play a track/value if the logical switch is on.

 One weird thing for my setup is the head tracker channels only work if set to use channels below ch 8 I think. So if it's set to start at ch 6 then 6, 7, and 8 show pan, tilt, and roll. If it set to channel 7 it only shows 2 of the head tracker ch's and so on. I might have the numbers wrong but I tested the problem on many channels and it was consistent. 

I know this isn't the head tilt alarm many people wanted like in orqa's (I think) but it's a work around until the software can have this feature added hopefully.