hd-zero/hdzero-goggle #70: Feature request: osd show arrow where to point patch antennas for best reception

Combinacijus, 2023-01-10:
(If this is not possible due to hardware limitations just close the issue)

# Problem
High gain patch antennas have narrow fov so how you point your head can have a big difference, and in long range you have limited patch of sky to fly in forward and back

# Solution
Add osd element which points to the direction where you should point your head for best reception

This would allow:
- Long range pilots to optimize their head orientation for best signal reception
- Long range pilot to fly in any 360 direction at long range, because they can turn their head with the drone and always have range of patch antennas

# Hardware capatibities needed
- Drone should have GOS receiver
- Drone GPS data should get back to googles. This part I'm not sure how it would work, maybe from elrs telemetry from radio or something like that
- Googles need absolute positioning for yaw and pitch axis (or relative but with small drift during flight like 2deg or so)
- Googles doesn't need GPS because you could assume pilot position to be same as first arm position, at longer range angle error would be minimal

TwistedFPV, 2023-02-26:
Bump
This is a great idea!

ligenxxxx, 2024-06-06:
Home direction?