hd-zero/hdzero-goggle #71: More documentation for developers?

Combinacijus, 2023-01-10:
Is there more documentation for developer?

Like high level block diagram how hardware connected what sensors available. What's possible with video postprocessing. What kind of OS is running. How to test custom firmware. Some example code for some simple projects like reading sensor data and displaying in terminal, osd or whatever sensible? Other info to help to start on basics without needing to read through all source code to understand what does what

I think something like that would lower entry barrier for developers to join the development

a-di-ez, 2023-02-06:
@bkleiner Is there a tentative date for the implementation of this, as it will speed up the work on other issues many times over?

bkleiner, 2023-02-06:
Well, since you directly ping'd me...
Generally speaking i do agree that more developer documentation would be helpful to get people started.

However i think expectations need to be adjusted a little here. Divimath (who do _not_ employ me btw) has _not_ sold you a piece of educational hardware. This is not a raspberry pi.
Its purpose is not to teach you how to work a embedded system. 
So i dont think its reasonable to expect them come up with example code on how to work a single given component. 

I do agree that we should we try document things as we figure them out or as they are communicated by divimath, but this has to very much be an ongoing effort and not something that comes with a fixed implementation date.

a-di-ez, 2023-02-06:
@bkleiner unfortunately, but you are absolutely right, thanks

closed 2024-06-07 by ligenxxxx