hd-zero/hdzero-goggle #84: Not Recording DVR

Proskate, 2023-01-13:
REC icon shows up, but DVR is not saved. File format, AUTO/Manual doesn't matter. Tried Latest stable FW and compiled current GH version, but still no recordings. In manual mode on current GH version REC icon blinks only once. SanDisk Ultra 32GB and it's worth to notice that few first flights were recorded.

bkleiner, 2023-01-13:
i had [made an error](https://github.com/hd-zero/hdzero-goggle/pull/85) when importing the latest record app into the version. could you please update from GH, format the sdcard within the goggles and try again?

bartekmanu, 2023-01-13:
I have similar problem but on Rev_06012023 version. No recording. I'm not sure but for sure it's happen when I plug off LIPO from quad when battery on osd blinking, something around 3.6-3.7V.

Also ( maybe I should open new issue ) goggles immediately trying restart after plug off LIPO from quad. I said `trying` cuz there is HDZero img with blinking dots so it's loading process which never end :/ maybe this two things are connected

Proskate, 2023-01-13:
@bkleiner Seems working on bench, tommorow will test it little bit more. Thank You. 

It's also worth to notice that to have auto record I had to switch to manual and go back to auto.

bkleiner, 2023-01-13:
@bartekmanu that seems like another issue altogether and like one that is probably best communicated to divimath support directly.

@Proskate good. let use know how it goes.

Proskate, 2023-01-15:
@bkleiner Recording works fine now, have every single one DVR from yesterday.

Proskate, 2023-01-15:
@bkleiner Ok, today found out that after formatting SD DVR records fine, but after plugging into PC and back to Goggles won't record. Will try to do same thing but with different PC to see if it's some kind of Windows issue

bartekmanu, 2023-01-15:
I also noticed this, but I checked it thoroughly and `Rev_22122022` fw doesn't have that problem. I guess that behavior appears on `Rev 06012023`

AlexandreBonneau, 2023-01-22:
Just to add; using the latest available firmware `Rev 06012023`, I also got the same behavior; the first few flights were recorded, but not anymore.

AlexandreBonneau, 2023-01-24:
To add more info on the bug, it seems the problem comes from the SD card being mounted read-only.
At least that's my guess is happening on the goggles, since when I insert the card in my computer (running Linux), I get the following kernel messages:
```
[65778.322421] usb 3-2: new high-speed USB device number 6 using xhci_hcd
[65778.480975] usb 3-2: New USB device found, idVendor=14cd, idProduct=1212, bcdDevice= 1.00
[65778.480982] usb 3-2: New USB device strings: Mfr=1, Product=3, SerialNumber=2
[65778.480984] usb 3-2: Product: Mass Storage Device
[65778.480986] usb 3-2: Manufacturer: Generic
[65778.480987] usb 3-2: SerialNumber: 999999999999
[65778.481583] usb-storage 3-2:1.0: USB Mass Storage device detected
[65778.481716] scsi host3: usb-storage 3-2:1.0
[65779.494927] scsi 3:0:0:0: Direct-Access     Mass     Storage Device   1.00 PQ: 0 ANSI: 0 CCS
[65779.495230] sd 3:0:0:0: Attached scsi generic sg2 type 0
[65779.724083] sd 3:0:0:0: [sdc] 487325696 512-byte logical blocks: (250 GB/232 GiB)
[65779.724173] sd 3:0:0:0: [sdc] Write Protect is off
[65779.724176] sd 3:0:0:0: [sdc] Mode Sense: 03 00 00 00
[65779.724260] sd 3:0:0:0: [sdc] No Caching mode page found
[65779.724262] sd 3:0:0:0: [sdc] Assuming drive cache: write through
[65779.725838]  sdc: sdc1 sdc2
[65779.726010] sd 3:0:0:0: [sdc] Attached SCSI removable disk
[65785.108260] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[65799.277625] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277631] FAT-fs (sdc1): Filesystem has been set read-only
[65799.277636] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277638] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277639] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277641] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277642] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277643] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277644] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277645] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277646] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
```

Looks like FAT32 does not like when you switch off or unplug your battery from the goggles, hence creating the usual `not properly unmounted` error message.

One solution perhaps could be to use a newer filesystem for the SD Card? Either exfat for windows users, or f2fs/ext4/btrfs/xfs/etc. for Linux users?

AlexandreBonneau, 2023-01-25:
For info, I was able to 'fix' the SD card by running fsck:
```
# fsck /dev/sdc1
fsck from util-linux 2.38.1
fsck.fat 4.2 (2021-01-31)
There are differences between boot sector and its backup.
This is mostly harmless. Differences: (offset:original/backup)
  65:01/00
1) Copy original to backup
2) Copy backup to original
3) No action
[123?q]? 1
FATs differ but appear to be intact.
1) Use first FAT
2) Use second FAT
[12?q]? 1
/isp0_0_0_0_ctx_saved.bin
  Contains a free cluster (728715). Assuming EOF.
/isp0_0_0_0_ctx_saved.bin
  File size is 40472 bytes, cluster chain length is 0 bytes.
  Truncating file to 0 bytes.
Reclaimed 2 unused clusters (65536 bytes).
Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
[12?q]? 2
Free cluster summary wrong (1382323 vs. really 1382325)
1) Correct
2) Don't correct
[12?q]? 1

*** Filesystem was changed ***
The changes have not yet been written, you can still choose to leave the
filesystem unmodified:
1) Write changes
2) Leave filesystem unchanged
[12?q]? 1
/dev/sdc1: 87 files, 714313/2096638 clusters
```

Then I was able to fly 4 batteries and get the first 2 recordings.
Now when I plug the SD card again in my computer, I again see the usual errors:
```[95250.360108] usb-storage 3-2:1.0: USB Mass Storage device detected
[95250.360234] scsi host3: usb-storage 3-2:1.0
[95251.377167] scsi 3:0:0:0: Direct-Access     Mass     Storage Device   1.00 PQ: 0 ANSI: 0 CCS
[95251.377404] sd 3:0:0:0: Attached scsi generic sg2 type 0
[95251.662897] sd 3:0:0:0: [sdc] 487325696 512-byte logical blocks: (250 GB/232 GiB)
[95251.662998] sd 3:0:0:0: [sdc] Write Protect is off
[95251.663002] sd 3:0:0:0: [sdc] Mode Sense: 03 00 00 00
[95251.663090] sd 3:0:0:0: [sdc] No Caching mode page found
[95251.663093] sd 3:0:0:0: [sdc] Assuming drive cache: write through
[95251.664859]  sdc: sdc1 sdc2
[95251.665065] sd 3:0:0:0: [sdc] Attached SCSI removable disk
[95253.976447] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[95257.202653] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202659] FAT-fs (sdc1): Filesystem has been set read-only
[95257.202664] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202666] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202667] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202668] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202669] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202670] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202671] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202672] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202673] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
```

I made sure I used the goggles switch button each time, instead of unplugging the barrel connector, while I doubt it would change anything.
Somehow, doing this with the HDO2 never lead to read-only SD card; perhaps Fatshark did run an equivalent of fsck on boot?

bkleiner, 2023-02-03:
@AlexandreBonneau could you me a solid on open a new ticket with the information about the filesystem corruption you compiled there? i agree. running fsck during boot/before mount is probably what ultimately needs to happen.

For any other issues that might pop-up, let's please open tickets dedicated tickets, with one ticket per issue as well.
These catch-all "feature X does not work" tickets become convoluted quick.

closed 2023-02-03 by bkleiner

AlexandreBonneau, 2023-02-03:
Well, I posted the reason (I'm guessing) why the DVR was not recording, as well as a workaround.
I think this is related to this current issue, and did not warrant another to be created, but I still [did](https://github.com/hd-zero/hdzero-goggle/issues/151) since you asked :)

bkleiner, 2023-02-03:
> Well, I posted the reason (I'm guessing) why the DVR was not recording, as well as a workaround. I think this is related to this current issue, and did not warrant another to be created, but I still [did](https://github.com/hd-zero/hdzero-goggle/issues/151) since you asked :)

i am fairly certain there are multiple issues causing the dvr recording to fail, the filesystem corruption being one of them.
i would like to record each issue in its own ticket so we can make sure none get lost in the noise and so that we can tick them off one by one.
i fully agree with your assessment of the filesystem situation. it will just be a little longer until we are in a position where we can implement your suggestion. (we need a way of patching the rootfs first)
i don't want your research to get lost until then.