hd-zero/hdzero-goggle #96: Feature Request: Tag good DVR clips

evilC, 2023-01-18:
When flying, a lot of clips tend to accumulate on your DVR, most of which you are probably not interested in keeping.  
Typically, when I get home, I have to review the DVR files to see what I want to keep.  
Sure, you could delete stuff as you go, but that's a bit time-consuming.  
It would be really cool if there were some way to mark a DVR file as "Hot" - eg you have a flight and did something cool, so after the flight, you go into the DVR menu, select the last file, and select a menu item to tag it as "Hot".  
Then, when you get home, you know that all you have to do is copy off the "Hot" clips and can ignore the rest.  

Maybe a simple solution would be that you can go into the DVR thumbnail view, and press the record button while a clip is highlighted, and it moves that video into a "Hot" subfolder, or maybe renames the file to put a `Hot_` prefix at the start of the filename

bkleiner, 2023-01-19:
i think this is a super cool idea, i wanted to re-work the right button anyway, so i took the opportunity.
this changed how the dvr record is triggered, so i would appreciate if you could test if that is still working as intended.

evilC, 2023-01-19:
Wow! didn't expect that so fast!
I have not had a chance to test yet, but a quick glance at the code says to me that there could be a potential issue - I made a comment on the code.

I am no C expert, and unfamiliar with the code base, but what happens? is `hdz_123` renamed to `hot_hdz_123` or just `hot_123`?  
If the former, what happens if you tag the same file twice? Does it become `hot_hot_hdz_123`?  

Furthermore - with this feature being implemented, I would say that #97 becomes a little more complicated - I think we should have a "Delete all recordings except hot recordings" option. Also, wrt #98 , this current implementation maybe precludes that? What's the plan here? If that is to be implemented, would the "Mark as hot" option become an entry in the context menu?  

If you wanna discuss some of these ideas without endless posts in the issues forum, pls feel free to hit me up in the HDZ discord channel

bkleiner, 2023-01-19:
> I am no C expert, and unfamiliar with the code base, but what happens? is `hdz_123` renamed to `hot_hdz_123` or just `hot_123`? If the former, what happens if you tag the same file twice? Does it become `hot_hot_hdz_123`?

let's discuss this in the PR

> Furthermore - with this feature being implemented, I would say that #97 becomes a little more complicated - I think we should have a "Delete all recordings except hot recordings" option. 

i don't see really see how it _nessacitates_ that, but such a function can be considered.

> Also, wrt #98 , this current implementation maybe precludes that? What's the plan here? If that is to be implemented, would the "Mark as hot" option become an entry in the context menu?

Why would it preclude that? really that PR offers a basis for further extension.
Previously the right button could _only_ be used while on the video screen.
Yes, the mark as hot can be moved to the context menu if that menu were to be implement.

commit 9194b741 (2023-01-23): add function to mark file as "hot" fixes #96
closed 2023-01-23 by bkleiner