hd-zero/hdzero-goggle #108: The Foxeer Wildfire analog module output a bad image

AlexandreBonneau, 2023-01-22:
Using me Fatshark HDO2 with the Foxeer Wildfire analog module, a luminier omni antenna and a sniper patch antenna, I got very good results and my analog image is stable and clear.

However when using the same Foxeer Wildfire analog module with the same antenna, the same orientation and the same flight conditions, I see a pretty bad analog signal, with lots of break ups, lots of rolling frames (that I never experienced before with the HDO2).

Help!

*(I can provide a flight recording to the devs if needed)*

bkleiner, 2023-01-23:
Could you provide the videos here? just so we have them on record.

AlexandreBonneau, 2023-01-24:
> Could you provide the videos here? just so we have them on record.

Would you have a way to send this privately?

bkleiner, 2023-01-24:
I would like to have the behavior recorded here, so we can match other cases and allow other people to get an idea on what we are looking at. If you have privacy concerns with your current footage, please try to produce some footage that does not reveal anything sensitive.

AlexandreBonneau, 2023-01-27:
The video with bad signal can be downloaded on [this link](https://mega.nz/file/CvIHxYrQ#sCnpWuC88MS_OsZaY66XrKDADqirc4l4Ruic60aDoaA).

bkleiner, 2023-02-18:
i have acquired a wildfire module to try and work on this, however i have to go to quite extreme lengths to produce anything even close to what you are seeing. Your incoming video signal must be quite bad.
Please verify your antennas & other involved hardware. The signal should not be this bad just flying in what appears to be your backyard.

AlexandreBonneau, 2023-02-19:
Hello @bkleiner, thanks for investing this problem.
To add to the initial issue, I've tested again and here are my findings:
- Using the same module with the same antennas in my HDO2, I get a perfect signal (at the same spot, using the same flight path)
- It's subtle, but when I sit one meter toward the door (so approx. 50cm to the opening, still inside the house), I get a better signal
- I've tried in an open field, and the signal is 'ok' (not very good, just good enough)
- My antennas are pretty good too; one 19cm omni luminier, and one truerc sniper patch antenna

So from what I saw, it seems going through only one cinder-block wall deteriorate the signal _a lot_ when using the HDZero analog module.
For comparison when I'm in the middle of the house (at least 2 thick 20cm cinder-block walls), I can fly around the house without any problem, specially no rolling image which are the most disturbing glitch.

I've recorded additional footage comparing the same setup HDO2/HDZero googles, if needed, but the buggy behavior is already linked above.

My empiric conclusion is that somehow the analog module does not correctly manage the signals when behind walls, while it should be able to (since the HDO2 can).

I can test more if you need anything

bkleiner, 2023-02-20:
> So from what I saw, it seems going through only one cinder-block wall deteriorate the signal _a lot_ when using the HDZero analog module (also, when I fly around one single metallic (6x6cm) vertical square tube, the signal takes a hit too). For comparison when I'm in the middle of the house (from 2 to 4 thick 20cm cinder-block walls), I can fly around the house without any problem, specially no rolling image which are the most disturbing glitch.

Curios. i have to place a whoop at 25mw behind a concrete wall _and cover it with a baking tray_ to produce a image roll.

> My empiric conclusion is that somehow the analog module does not correctly manage the signals when behind walls, while it should be able to (since the HDO2 can).

Do note the RF handling is done in the wildfire, the goggles merely accept the video signal. If you take a look at the hdz "analog module", you will notice that there are no active components, it's simply traces. It has no control over RF processing.

It does however have _some_ control over the strategies used to try to hang on to this video signal (hardest part about analog video is tracking the sync pulses).

> I've recorded additional footage comparing the same setup HDO2/HDZero googles, if needed, but the buggy behavior is already linked above.

A side-to-side would could be useful. 
In your first clip the wildfires own sync detection seems to be going out constantly (the RF indicator keeps disappearing), is that consistent between your tests?

AlexandreBonneau, 2023-04-07:
Yes, it is.
I've tried in an open field to minimize any interference and bouncing signal, and somehow, with that quad specifically, the signal is somehow pretty bad at about 50 meters away.
However I can still see that same signal correctly when using the HDO2.

What's interesting is that from my testing, all my other analog quad works the same between the hdzero and HDO2 google.
This one, an HGLRC BNF Racewhoop25, seems to be the only one that yield such bad signal on the HDZero googles only.
I do not have another HGLRC analog VTX to test on another quad.

I'll try to setup a side by side comparison if I can.