hd-zero/hdzero-goggle #115: Bug: Microphone audio problem when analog module connected

uPz3TT3r, 2023-01-23:
I use an external microphone, but when the analog modul is mounted, it destroys the audio.
I made a video to show the problem:
https://www.youtube.com/watch?v=PS3IaPrHgcI

Used HDZero Goggle with Rapidfire and external Mic.
Same problem if you fly analog but want microphone audio only, the analog sound is also written into audio file. 
Please change to use only 1 input source.

uPz3TT3r, 2023-01-23:
I forgot that it is possible to turn off audio on rapidfire modul itself. If you turn of audio there, it is ok. 

Dont know if it really is a bug in that way. What if analog modules dont support to turn off audio?

bkleiner, 2023-01-24:
Not exactly a bug, no. Lets please convert this into a Feature Request to allow for selecting the analog audio source. (maybe re-open for clarity)

uPz3TT3r, 2023-01-24:
I go with your advise to change this to feature request. I dont know if for example true-d modules can turn off audio, so another audio source for analog would be great (and to activate only the selected source)

cougman, 2023-11-18:
I'm using Rapidfire as well , with the Rush Tank with onboard mic .For me turning off the audio , does just that ...I'm not able to hear anything after selecting this . This has no relation to the current bug that exists.  The mic is only enabled after selecting the expansion as source, and not on startup.