hd-zero/hdzero-goggle #116: Feature request: Support timestamp in DVR metadata

Wikiwix, 2023-01-23:
As discussed by @rquellet in [this YouTube video](https://www.youtube.com/watch?v=p4qXiQrtCkU&t=443s) there is a RTC in the goggles.
Having this information in the DVR files is essential for me and allows easier management of flight footage.

razzeee, 2023-01-23:
related to https://github.com/hd-zero/hdzero-goggle/issues/42

evilC, 2023-01-27:
Out of curiosity, the video says there is a connector, but it's not too clear exactly what the connector is.
Is it a cell holder? ie can we just pop a battery in there, or is it a separate cell holder and what is on the PCB is just a pin header to connect the cell holder?

Wikiwix, 2023-01-27:
@evilC Ryan points at the connector. Looks like some JST connector.
If you google "*RTC battery*" , you will find examples of CR3032 with a plug. All I saw seemed to use the same connector so I guess this is kind of *the standard*.

Master92, 2023-11-11:
Isn't this feature implemented? Maybe you should consider closing this and related issues that have been implemented by now 😉

Wikiwix, 2023-11-11:
@Master92 This particular one is not fully implemented AFAIK (though I am still on 9.0). The file gets creation and edit timestamps, but the mp4 does not get the typical metadata yet.
See [this comment](https://github.com/hd-zero/hdzero-goggle/issues/123#issuecomment-1484204361) for some research I did a while ago.
Closed the other issue I created as completed though.

Master92, 2023-11-11:
Oh okay maybe I've mistaken the file creation date - which is set - for the video metadata.
I'll have a look at that when I've got some time. Your approach seems promising and I'll update as soon as I've come around to trying it out.

PR #379 (2024-01-05): Add date-based naming scheme for DVR files and add video metadata
This PR allows users with an active RTC to chose between the conventional naming scheme (i.e. hdz_000, hdz_001 etc.) and a date-based one (i.e. 2023-11-24T13_34_37).
When no rtc battery is present, the user can not change the naming scheme. On top of that, it automatically falls back to the conventional scheme as default.

Further, I added some metadata to the DVR files i.e. a title (currently the file name) and - for mp4 - the date it was created.
Over the course of my research and the implementation, I found out that the ts co [... cut]
closed 2024-01-05 by ligenxxxx