hd-zero/hdzero-goggle #118: Feature Request: Show active Channel in HDZero OSD permanently

uPz3TT3r, 2023-01-24:
If you change a channel, the active channel is shown in OSD for some seconds, but goes away after that.
In Conditions with more pilots it is nice to know on which channel the goggle is set at the moment. It also helps in directly be able to see and fix a missconfiguration.

timfredriksson, 2023-01-29:
That sounds unnecessary as long as you can choose if you want it or not in the setting it seems fine

aviphysics, 2023-01-30:
What if it showed channel for a couple seconds whenever the goggle OSD was turned on, so you could always check your channel with at most a quick double click of the left button?

uPz3TT3r, 2023-01-31:
> 

@timfredriksson that would be perfect: a setting area in menü where you can turn off/on every OSD setting that you want permantly. Fan, Channel, Battery, Signal, Lock etc. etc. 

That would be nice.