hd-zero/hdzero-goggle #123: Enable RTC functionality

Wikiwix, 2023-01-26:
This issue is a container for making use of the RTC.

### Situation statement

A first step is to allow setting the datetime (and potentially the timezone) and having a secure way to determine a proper datetime is set. This is not a given as there is some tinkering involved to install an RTC battery (see [this video by @rquellet ](https://www.youtube.com/watch?v=p4qXiQrtCkU&t=443s) for details).

### Potential use cases

* DVR recordings can contain metadata with the recording datetime #116 
* DVR files could potentially contain a timestamp #42 
* The OSD menu could show the time (not sure about the value)

### Implementation suggestion

* Provide a settings entry to set time
  * This should allow setting
    * year, month and date
    * hours and minutes
    * Optionally set a time zone (otherwise the metadata will only be correct if the set datetime is given in UTC)
  * The setting should have an asterik (`*`) like the WiFi setting currently has. The footer should contain the warning
    > Time will be reset on reboot unless RTC battery is installed!

    If there is a simple way to check whether an RTC battery is installed, the asterik could be shown conditionally with a message like

    >  Time will be reset on reboot because no RTC battery is installed!
  * Currently that menu entry fits probably best in `Record Options` (already quite crowded) or `Firmware` (this could be renamed to `System` to be a little more generic)
* Features that want to use the datetime should check for the following precondition (pseudo code) to ensure a plausible datetime is configured
  ```python
   DateTime.now() >= thresholdDateTime
  ```
  `thresholdDateTime` has to fulfil the following conditions (`2023-01-01T00:00:00Z` might match them)
     * The lower bound for `thresholdDateTime` should be a date older than the day this feature ships
     * The upper bound for `thresholdDateTime` should be at least some days after the datetime the RTC starts at after loosing power
     * The lower and upper bound together ensure that metadata is only set if a plausible datetime is configured

bkleiner, 2023-01-28:
Lovely! Thank you.
i like that there is pre-planning for informing the user. feels like a good compromise to me.
Do you think you would be able to put forward a PR implementing this?

rquellet, 2023-01-29:
Here's an interesting workaround: what if we use ELRS backpack to push the RTC timestamp from radio to hdzero goggles?

bkleiner, 2023-01-29:
> Here's an interesting workaround: what if we use ELRS backpack to push the RTC timestamp from radio to hdzero goggles?

does not change much in actuality, as this is not something that we can rely on being preset.
so all the checks outlined above would still need to be in-place.
manual entry is just one possible way of obtaining the timestamp, elrs, wifi, etc are all possible inputs to this system.

SumolX, 2023-03-10:
I really like the layout of this plan.  My first goal was to have the RTC driver implemented prior to tackling this proposal.  I would like to break it down into a few phases.

Phase 1: RTC Driver Support (see pull request):
https://github.com/hd-zero/hdzero-goggle/pull/191

Phase 2: RTC Driver Enhancements:
New method which will generate a filesystem safe naming convention similar to ISO 8601: 
    YYYYMMDDTHHMMSS
    20230310T011735

New method which will generate a pretty naming convention for use in the OSD:
    Fri Mar 10 01:17:35 2023   or    2023/03/10 - 01:17:35

New method which can report the detection of altered RTC clock:
   return year != 1970 && month != 1 && day != 1;

Phase 3: Display date/time within OSD menu via a new submenu:
  Create new submenu for manipulating RTC date/time
  Have the option to display the date/time if the RTC clock contains an non EPOCH date onto the OSD. (Use * to denote why it maybe disabled)
  Provide the ability to modify the RTC Clock via this same menu: 6 fields containing their respective attributes for year/month/day/hour/min/sec

Phase 4: Alter naming convention of DRV recordings:
  If the RTC clock contains an non EPOCH date, then automatically reference the filesystem safe naming convention method and prefix the formatted timestamp onto the name of the recorded file.

Phase 5 Investigate the use of metadata for timestamping and provide possible solutions... like the one described above.

Wikiwix, 2023-03-13:
@SumolX Happy to hear that you are working on this. And I like your implementation sketch overall 👍🏻.

Nevertheless I have some suggestions/remarks  (I hope that this is welcome):

### General stuff

#### Time zones

* I suspect the RTC hardware does not care for time zones and the OS would have to take care of this (then the RTC should contain UTC).  
  So if timezones should be considered it is a concern from Phase 2 onwards
* Why would timezones be important:
  * the timestamps in video metadata contains timezone (for `CreationDate`) and expected to be UTC for all other fields (see [Stack Exchange](https://superuser.com/a/1285932) and the linked docs)
  * For the file timestamp it technically does not matter as FAT is used:
    > The FAT file system stores time values based on the local time of the computer.
    from [Microsoft's docs](https://learn.microsoft.com/en-us/windows/win32/sysinfo/file-times)
  * For anyone in a timezone significantly away form GMT, setting the clock to UTC is no nice workaround I guess
    * due to it being shown in the UI
    * and it being not that intuitive

#### Order of Phases

I personally would moved some of the changes to a later stage:

* Showing the clock in the OSD is something I would have tackled *after* Phase 5. I personally believe that this is not that important for most. If one wants to see the time one can use the Betaflight OSD ¯\_(ツ)_/¯.  
  This would save an  (arguably simple) method in Phase 2 as well
* I would implement the UI based date/time config to after Phase 4. Adding the txt file based on https://github.com/hd-zero/hdzero-goggle/pull/191 seems feasible for most who care enough (at least temporarily).  
  And I guess that modification menu item is the most complex part of the whole plan and not absolutely necessary to start

### Remarks regarding specific phases

#### Phase 2

##### filesystem safe naming convention

I would use a slightly different filesystem naming convention: `YYYY-MM-DDTHH_MM_SS` I have two reasons for that:

* my proposition is a lot more glanceable for a human
* if someone needs to (for whatever reason) extract the timestamp from the filename, a simple `sed 's/_/:/g'` will produce an easy to parse ISO 8601 timestamp

##### pretty naming convention

If you choose the second option, I would make it a little more ISO-8601-like and loose the `-` as the date-time-separator:

`2023-03-10 01:17:35` This is exactly the format Betaflight uses in the OSD

My motivation for this is internationalization. As long as there are no different localizations there will be just one type of "pretty" date. An ISO-8601 date would reduce ambiguity for anyone outside the US 🙈

Your first option `Fri Mar 10 01:17:35 2023` seems okay as well, but:

* In these Goggles, do we really need the day of the week?
* The format uses a lot of space without adding any "real" benefit
* If the Goggle menus would ever become localized, this function would need to change, with the other suggestion that would be optional

#### Phase 5

* See [Stack Exchange](https://superuser.com/a/1285932) and the linked docs for the relevant tags
* I think I saw ffmpeg being used. Based on Stack Overflow it can take care of setting *all* relevant date/time info using one parameter: https://stackoverflow.com/a/60534938

SumolX, 2023-03-14:
Thank you for the feedback.  Feel free to reach out to me on discord.  This will allow us hash this out in real-time vs going back and forth via comments.

Wikiwix, 2023-03-26:
@SumolX asked me to do some research on setting metadata for the recordings.

Unfortunately **I failed** to create a firmware which sets the desired timestamps with a dummy timestamp.  
But I will share my results, so someone with any experience in C can have a look:

* Using the ffmpeg cli the command to set the desired metadata is really simple:
  ```sh
   ffmpeg -i  hdz_009.mp4 -c copy -map 0 -metadata creation_time="2023-02-03 04:05:06"  test.mp4
  ```
  successfully sets all relevant metadata timestamps to the specified time  

* `exiftool` (and to some extend `ffprobe`) can be used to check on the available metadata:
  ```sh
  # Add the commented out stuff to only see the relevant 6 metadata fields
  exiftool -g -sort hdz_009.mp4 # -G | grep Date

  # shows the `creation_time` metadata element
  ffprobe hdz_009.mp4
  ```
* My attempt to prove the functionality in the goggles was based on [this stackoverflow](https://stackoverflow.com/questions/45050177/adding-metadata-informations-with-ffmpeg)
  ```diff
  diff --git a/src/record/ffpack.c b/src/record/ffpack.c
  index 757f7bc..2a04c00 100755
  --- a/src/record/ffpack.c
  +++ b/src/record/ffpack.c
  @@ -323,6 +323,8 @@ int ffpack_start(FFPack_t* ff)
       int ret = 0;
   
       //ffpack_dumpTimebase(ff);
  +    av_dict_set(&ff->ofmtContext->metadata, "creation_time", "2023-02-03 04:05:06", 0);
  +    av_dict_set(&ff->ofmtContext->metadata, "title", "testtest", 0);
   
       ret = avformat_write_header(ff->ofmtContext, NULL);
       if (ret < 0)
  ```
  and I also tried the solution of [that stackoverflow](https://stackoverflow.com/a/32080235):
  ```diff
  diff --git a/src/record/ffpack.c b/src/record/ffpack.c
  index 757f7bc..0e41876 100755
  --- a/src/record/ffpack.c
  +++ b/src/record/ffpack.c
  @@ -324,6 +324,11 @@ int ffpack_start(FFPack_t* ff)
  
       //ffpack_dumpTimebase(ff);
  
  +    AVDictionary *pMetaData = NULL;
  +    av_dict_set(&pMetaData, "creation_time", "2023-02-03 04:05:06", 0);
  +    av_dict_set(&pMetaData, "title", "testtest", 0);
  +    ff->ofmtContext = pMetaData;
  +
       ret = avformat_write_header(ff->ofmtContext, NULL);
       if (ret < 0)
       {
  ```
  In both cases the files get written as expected, but no extra metadata is written (not even the `title`)
  My theory for failing is one of
  * I made a mistake using C (I have last worked with pointers and memory allocation ~10 years ago…)
  * [This comment](https://stackoverflow.com/questions/32078807/metadata-is-not-showing-ffmpeg-c#comment79330317_32080235) applies. [That documentation](https://wiki.multimedia.cx/index.php?title=FFmpeg_Metadata) might be relevant in that case
  * I failed to make the change in the right location

I am happy to answer further questions I can help with :)

And I am glad if someone can point out the mistake(s) I made 🙊

Wikiwix, 2023-11-11:
Besides https://github.com/hd-zero/hdzero-goggle/issues/116 RTC functionality is available

closed 2023-11-11 by Wikiwix