hd-zero/hdzero-goggle #124: UI sliders move in opposite direction to what one would expect

evilC, 2023-01-27:
Within the UI, the convention for the jog wheel seems to be clockwise = next, anti-clockwise = prev.  
This makes logical sense - in FPV we call a clockwise prop a "right", and in general in UIs etc, clockwise is next (right / down) and anti-clockwise is previous (left / up)
However, for the sliders in the HDZero UI, the opposite is true - clockwise moves the bar to the left, anti-clockwise moves the bar to the right.

IMHO this should be reversed - turning the wheel clockwise should move the slider bar to the right

bkleiner, 2023-01-27:
These things seem to come up in every project that involves a control-scheme i have ever worked on.
There will always be people that perceive whichever direction you choose as "incorrect". Trying to argue one direction is incorrect is futile. People seem to not even be able to reach consensus on which direction mouse wheel scrolling is "correct" (apple default setting). 

As for it being inconsistent, i feel this is very much up for debate as well, moving the dial clockwise moves through the entries top to bottom, "down" so to speak. 
Applying the same rotation to a slider reduces the value, that checks out, imho.

bkleiner, 2023-01-27:
Yeah, playing with it some more, sorry, i am closing this one as matter of opinion.
Nothing is objectively wrong here, changing it will just be perceived as "wrong" by somebody else.

closed 2023-01-27 by bkleiner

evilC, 2023-01-27:
> These things seem to come up in every project that involves a control-scheme i have ever worked on. There will always be people that perceive whichever direction you choose as "incorrect". Trying to [...]

When in the Playback menu, scrolling clockwise moves the selection to the right.