hd-zero/hdzero-goggle #126: Bug / Feature request:  Voltage warning option in Auto mode broken, needs to be replaced with "total voltage"

evilC, 2023-01-27:
"Auto Mode" in this post refers to the legacy (Pre #122 ) mode for the Power menu, where it guesses the cell count, and gives you a low voltage warning based upon the value of `Guessed Cell Count` x `Minimum Cell Voltage`

When in Auto mode, the warning voltage is set on a per cell basis (ie it allows values in the range 2.8 ... 4.2)
Cell count is inaccurate in Auto mode (It misidentifies a 4S pack as 3S if your cell voltage is at 3v), so we should not be basing the warning on `Cell Voltage` when in Auto mode.

When in Auto mode, the `Cell Voltage` slider should disappear and be replaced with and option named something like `Voltage Alarm` and it should allow values from the minimum supported voltage (7v) up to some sane maximum amount.

The new Manual Cell Count Mode should keep using the `Cell Voltage` setting as it currently is

bkleiner, 2023-01-27:
Calling this method of determining a nominal voltage "broken" feels a little over-reaching, considering how many devices and tools in the FPV world use techniques similar tho it to automatically determine a nominal voltage.

Your proposal makes the "Auto" option not "Auto" anymore. If i were to switch from a 4s to 2s lipo i would have to adjust the Voltage manually, making it a worse version of the "Manual" cell count mode.

evilC, 2023-01-28:
> Calling this method of determining a nominal voltage "broken" feels a little over-reaching, considering how many devices and tools in the FPV world use techniques similar to it to automatically determine a nominal voltage.
> 
> Your proposal makes the "Auto" option not "Auto" anymore. If i were to switch from a 4s to 2s lipo i would have to adjust the Voltage manually, making it a worse version of the "Manual" cell count mode.

With it how it currently stands, if you leave it on Auto and have `Cell Voltage` set to 2.8v, then power up your goggles for a flight with a 4S pack at 2.8v per cell (11.2v), it misinterprets it as a 3S pack and you get no low voltage warning because it thinks that it's at 3.7v per cell.

You will most likely not get any warning whatsoever if you power up the goggles with a 4S pack at 3v per cell or lower - the goggles will most likely cut out way before that, because the low voltage warning would not show until you got to 2.1v per cell (Even though you told it to warn you at 2.8v per cell), by which time the pack won't be able to supply the amps to keep the goggles powered.

If, however, we implemented my suggestion and you had it in auto mode and set the `Total voltage` to 11.2v, then regardless of the state of the pack when you power up the goggles, it won't mis-interpret it. If you had the low voltage warning set to 11.2v and you plugged in a 3S pack when you normally fly a 4S pack, then that's you making the error, not the software and is entirely your fault. With the code as it stands, it's the goggles fault, because it did not warn you when the voltage got the the level you told it to warn you at (Because it mis-interpreted a 4S pack as a 3S pack)

bkleiner, 2023-01-28:
> With it how it currently stands, if you leave it on Auto and have `Cell Voltage` set to 2.8v, then power up your goggles for a flight with a 4S pack at 2.8v per cell (11.2v), it misinterprets it as [...]

By that line of argumentation, we should just remove the "Auto" mode altogether, because, again, if this proposal were to be implemented, it would just be a more in-convenient version of the the "Manual" mode.

I feel like on inherent problem here is that you assume a Li-Ion battery, where the code assumes a Li-Po. Maybe that should just be a switch or have the nominal voltage directly be configured.