hd-zero/hdzero-goggle #138: Feature request: No DVR kept if quad never armed

evilC, 2023-01-30:
Given I power up my quad
When I fiddle around with it, for example to adjust the camera focus
Then I do not want 10 of DVR filling up my SD card where I just bitch about how hard it is to focus the damned camera

This could probably be tied into whether you armed the quad or not, and I am guessing that the goggles are not aware of whether you are armed or not, but could this maybe be bodged by inspecting the state of the OSD element which tells you whether you are armed or not?

If we can detect armed status, then I see two ways to approach this:
1. Recording does not start until you arm
2. Recording starts when it currently starts, but if signal is lost before you arm, then the recording is deleted

I personally prefer (2) as sometimes you want some lead-in on your video before you arm - I often use that bit of the clip for the start of my videos while the music is ramping up.

Zuldan, 2023-01-30:
You are right, this feature is already available under the Backpack options. Here are the basic steps.

1. Update the ELRS firmware on your goggles
2. Bind the ELRS backpack on your radio to the ELRS backpack on your goggles 
3. Change DVR recording from auto to manual on your goggles 
4. Under backpack options on your goggles, configure the DVR recording options you want.

Now you will only get DVR recordings for your armed flights.

evilC, 2023-01-30:
Is it feasible to use the OSD method I mentioned to give this feature to non-backpack users?

bkleiner, 2023-01-30:
> Is it feasible to use the OSD method I mentioned to give this feature to non-backpack users?

Sounds very messy. we have more than one firmware to consider. All we get are the strings and positions, would be a huge guessing game and would break everytime the FWs change something. 

However, the arming state information could be embedded into the metadata packets that are sent alongside the OSD.

evilC, 2023-01-30:
OK, the OSD method sounds like a non-starter, but I still think it would be a worthwile addition to the metadata - not everyone flies ELRS (Or has backpack configured) and I think this would be a worthwile feature to have enabled out of the box.

evilC, 2023-02-01:
> 4\. Under backpack options on your goggles, configure the DVR recording options you want.

Took me ages to find this! For anyone else reading this, it's in the LUA script on your TX, not in the goggles

Anyway, IMHO this is not optimal.  
Doing this means that for any quad that does not have ELRS backpack set up, you have no auto-record.  

Also, it means you either need a separate switch on your TX to record, or tie it in to the arm switch, which makes it like approach (1) above (No recording until you armed).

IMHO approach (2) above (Start recording immediately, and if no arm during recording, delete on stop recording) is much more preferable

So TLDR, even with ELRS backpack, I would like to see an option like I mentioned implemented

Zuldan, 2023-02-05:
> Took me ages to find this! For anyone else reading this, it's in the LUA script on your TX, not in the goggles

You are 100% correct, my mistake. I've updated my post for the correct location.