hd-zero/hdzero-goggle #139: Feature request: Go straight to camera view on power up of goggles when "Last" mode is enabled

evilC, 2023-01-30:
Given `Auto Scan` is set to `Last`  
When I power up the goggles
Then it should not waste time loading and showing the menu before going to the camera view

Is this unavoidable? ie is it wasting any time by loading the menu? Can  we get a camera feed quicker by not loading / showing the menu?

bkleiner, 2023-01-30:
The menu is displayed while the divimath chips go through there "training sequence".
That is not avoidable and the scan screen is used as a loading screen of sorts.
i don't see how we can get rid off it without completely, as we need to display _something_ until the video is ready.

evilC, 2023-01-30:
If it's unavoidable, IMHO if we can, it would be better to leave it on the logo screen until the UI is ready for user interaction - as at least that has a waiting animation

bkleiner, 2023-01-30:
that would basically be #140, i have something along those lines in underway already ("loading screen" until the app is fully ready)

evilC, 2023-01-30:
OK, I'll close this one then

closed 2023-01-30 by evilC