hd-zero/hdzero-goggle #141: Feature request: Quad finder

evilC, 2023-01-30:
We should add a feature that helps you find a quad by examining the signal strength difference between left and right patch antennas

oldrootbeer, 2023-01-30:
Doesn't that mean leaving the vtx on? It'll fry / reduce power pretty quick.

a-di-ez, 2023-01-31:
@oldrootbeer Yes, if you leave it on all the time, but we can send a signal through the elrs communication channel to turn on the video link, for example, for 30 seconds to search, and then turn it off for 30 seconds to cool down, and so on through the cycle until we find the drone

evilC, 2023-01-31:
It's not my idea, it's been done before - it's just one of those things that other systems have but HDZero does not, so if we have it, it removes a reason to not go HDZero

ligenxxxx, 2024-06-07:
I plan to add a frequency analyzer instead of this.

closed 2024-06-07 by ligenxxxx