hd-zero/hdzero-goggle #145: Feature request: Setting to stretch the image from the analog module from 4:3 to 16:9

a-di-ez, 2023-01-31:
The tinywhoop cameras does not have settings to change it there, and the glasses do not have such an opportunity, so not the whole screen is used

wernight, 2023-01-31:
Caddx Baby Turtle (Larva X HD) supports 16:9 and 4:3 however it's native 16:9 (i.e. 4:3 is cropped). Also it has onboard DVR and footage in 16:9 is way better. The only issue is that HDZero Goggle doesn't support stretching the image to 16:9.

Flying with the wrong aspect ratio is just awful (and I don't mean rendering a 4:3 feed in 4:3 but stretching 16:9 into 4:3).

I do see that there is a difficulty to switch between 4:3 and 16:9 depending on the model possibly especially for people flying with both aspect ratios (rare).

a-di-ez, 2023-01-31:
> Caddx Baby Turtle (Larva X HD) supports 16:9 and 4:3 however it's native 16:9 (i.e. 4:3 is cropped). Also it has onboard DVR and footage in 16:9 is way better. The only issue is that HDZero Goggle [...]

Larwa is not a tinywhoop, for such sizes it is possible to adjust the aspect ratio, but I see no reason to use an analog if there is hdzero. The analog is relevant only for 65-75mm, because hdzero is heavy.
For tinywhoop, the optimal image quality and weight are only in runcam nano 3, but it does not have any settings.
I don't know how you stretch 4:3 to 16:9 so that it looks terrible, but for me it's not a critical difference

wernight, 2023-02-01:
To be clear I'm using an analog module on the HDZero. Yes I can change the aspect ratio and lose what I could see on the side to gain a bit on the horizontal resolution.

a-di-ez, 2023-02-01:
@wernight change aspect ratio in goggle or camera?

wernight, 2023-02-01:
@a-di-ez In the goggle so that if I set the analog camera to send 16:9 (received via TBS Fusion module) the HDZero Goggle would also render that in 16:9.

For now I've change the aspect ratio that the camera sends (i.e. I get a cropped view but a bit higher horizontal resolution). Noticed that the DVR recorded video is always 16:9 which is good.

wernight, 2023-02-01:
You mean change the VTX to something like a HDZero Whoop / Race? Sure that'd improve quality if it actually fits and works with the camera and DVR (which I doubt). In any case I have digital VTX on 5" but I got the analog module for smaller quads and cheaper builds.

Keeping analog either options (HDZero Goggle stretch the received analog feed to 16:9 vs cropping the analog VTX to 4:3) would have the same picture quality just "distributed" differently. I think the HDZero Goggles are also very popular with many analog users as I also noticed around me.

a-di-ez, 2023-02-02:
Yes, analog support was a decisive factor for me to get rid of hdo2

cynfewl, 2023-02-06:
I would also like the ability to display a 16:9 analogue image at the full 46 degree FOV of the goggles. There are cameras like the Caddx Ant Lite (not the FPV Cycle edition that is 4:3 native) that I can't currently use due to stretching, but would love to on lighter 65mm builds.

I really like the way the goggles handle the HDzero 16:9 vs 4:3 image where the goggles are always 16:9, but the image from the camera either fills up the screen (16:9) or is displayed with black columns either side (4:3). This means I don't have to set my goggles to a particular mode and can mix cameras at will.

If the analogue module behaved in the same way, that would be awesome, though if I had to choose a screen format for the analogue module to force full screen 16:9, that would also work.

HazzaHFPV, 2023-04-05:
16:9 mode for the analog module would be awesome. I do fly a lot of analog whoops still, one reason I bought the HDZero Goggles. Many of my whoops do have 16:9 which gets squished in the goggles unfortunately. So I'd be really happy for this mode.

dbloemhard, 2023-05-12:
I have gotten used to flying all my analog whoops on 16:9 (Skyzone 02s). Even the 4:3 cameras, when stretched to 16:9 look great. Its much easier to get used to than the other way around (4:3 stretched to 16:9). So I would be very grateful for this fix.

cynfewl, 2023-05-12:
I know there is a lot of work going on with the analogue interface at the moment and this is just one issue amongst many that probably have higher priority.

Does anyone know if these wishlist (but still important for those of us who want them) items are also being included in these works?

gonebdg, 2023-05-22:
How it's going?
Don't want to argue or talk much regarding 4:3 & 16:9
16:9 option is a must for hdz goggle.

slash128v6, 2023-09-12:
I would like this as well.

SupaflyFPV, 2024-06-26:
I'd really like to have 16:9 for the analog input. This could make them my main goggles. A lot of people fly 16:9 on analogue.

thespova, 2024-06-30:
I also primarily use 16:9 cameras on analog even on whoops. Telling people to switch to a different VTX is not a fix. There are lots of reasons to use analog even outside of whoops still and lots of people got the HDZ goggles because they also are supposed to include full analog support. I still feel like to date they still have a fairly limited implementation for analog however.

kyleireland, 2024-10-15:
Option to switch between 4:3 and 16:9 Analog.
Stretches 4:3 to 16:9.
16:9 as 16:9 rather than compressing it to 4:3.

kyleireland, 2024-10-19:
This is what it would possibly look like if you had the option of 16:9
4:3 with 16:9 clip.
BUT this is a 4:3 clip stretched.

https://github.com/user-attachments/assets/dcca1058-2ef6-41e6-a2c8-b1757314355e

I feel the 16:9 is needed as some analog cameras are 16:9 and they get compressed to 4:3

gerab, 2024-10-30:
Ok, this is ridiculous that we don't have it yet.

JSMiCh, 2024-10-30:
My old analog camera (foxeer T Rex) has sensor 16:9 and gives really nice image. It would be great to have the feature

etsach, 2024-10-30:
Just purchased HDZero goggles for the ability to do both HD and Analog, and with 16/9 signal having it squeezed to 4/3 is awful, it gives horrible motion sickness when doing barrel rolls.
This is a feature available on the cheapest possible goggles, it's not possible not to have it on the most expensive analog system.
Would be glad to help implementing it if I can contribute.

thespova, 2024-11-05:
How do we get this fix implemented? I wouldn't even consider analog supported with out the ability to support the higher end analog cameras. 

Is this something we should bug Divimath about? Do we just get enough people to comment here? Discord?

pfeerick, 2024-11-05:
I'm also surprised this still hasn't been implemented after all this time.
Or there to have at least been an explanation of not currently possible to
do/hard to do because of XYZ.

anilvermab, 2024-11-07:
At least add this to your 10.0 to-do list.

8xx8, 2025-01-08:
+1 to this feature request from me.. Was really surprised in a bad way that this is not supported.

thespova, 2025-01-08:
With all the interest and comments this seems to get it would be nice to get some comment from devs about the possibility of this happening.

I know everyone was saying analog is dead there for a bit but that's clearly not true. Analog still has a lot of use and interest. HDZ are supposed to be "the best you can do with analog" but not having common screen ratio that many of the best analog cameras are offered with is an issue.

nerdCopter, 2025-01-19:
PR https://github.com/hd-zero/hdzero-goggle/pull/472 now stretches analog to 16:9, but seemingly without option for 4:3 :(

binary here: https://github.com/hd-zero/hdzero-goggle/actions/runs/12760401014

thespova, 2025-01-19:
> PR [#472](https://github.com/hd-zero/hdzero-goggle/pull/472) now stretches analog to 16:9, but seemingly without option for 4:3 :(
> 
> binary here: https://github.com/hd-zero/hdzero-goggle/actions/runs/12760401014

Was that specifically supposed to add an option or change the aspect ratio of analog? It doesn't look like either was intended as far as I can tell.

nerdCopter, 2025-01-20:
> Was that specifically supposed to add an option or change the aspect ratio of analog? It doesn't look like either was intended as far as I can tell.

same conclusion here. i don't how it's happening, only that it is.

thespova, 2025-01-20:
I gave this a few test flights with my whoops and didnt notice any issues. Can confirm its full screen with the aircraft I tested and all of these where set to wide screen on the cam side (or just have not adjustable wide screen cams).

Maybe they are just changing to wide screen being default for analog? IDK how many cams are sold default wide vs 4:3 but maybe this makes more sense as a default?

nerdCopter, 2025-01-20:
i just did some testing, i used the right-button input = toggle-source, and once i go from Analog to HD then back to Analog, the screen is 4:3 again. so i'm a little confused at the result/function.

i have not tested how to achieve 16:9 again. 😆 🤕 

as far as two firmwares, i think such would be messy in the dev-ops arena.  one firmware makes more sense.

etsach, 2025-01-20:
Ok, so at least even if it’s bugged or stuck in 16:9, the hypothesis that
it’s impossible for hardware reasons can be dismissed now. That’s good!

cougman, 2025-01-24:
> option was merged: #473
>
> latest main build(s) found here: https://github.com/hd-zero/hdzero-goggle/actions?query=branch%3Amain+event%3Apush

I’ll give it a whirl first chance Cold weather has changed my lifestyle somewhat 😟

AndrijGorda, 2025-01-27:
I tried to install the artifact from here: https://github.com/hd-zero/hdzero-goggle/actions/runs/12948041847, but I don't know if it was a coincidence or it has an issue, but after repower, my goggles won't boot.  I made a recovery SD and, after a few attempts, was able to restore the system to try to install the 16:9 new feature once again, but this time from the latest artifact that I took here:
https://github.com/hd-zero/hdzero-goggle/actions/runs/12948041847/artifacts/2480128676
and _it works great_.

gerab, 2025-01-27:
I tried to install the artifact from here: https://github.com/hd-zero/hdzero-goggle/actions/runs/12948041847, but I don't know if it was a coincidence or it has an issue, but after repower, my goggles won't boot. I made a recovery SD and, after a few attempts, was able to restore the system to try to install the 16:9 new feature once again, but this time from the latest artifact that I took here:
https://github.com/hd-zero/hdzero-goggle/actions/runs/12948041847/artifacts/2480128676
and it works great.

nerdCopter, 2025-01-27:
> [...] I made a recovery SD and, after a few attempts, was able to restore [...]

i've had similar before, ; i was told simply to put one of the `*_VA.bin` file on SDcard root, it should repair without doing the entire recovery method.  Also, What i found missing from the user-manual (v1.9) about the "Emergency" recovery method is that the PhoenixCard tool reformats to exfat, but you still need to reformat into fat32 for the `_RX`/`_VA` part to work.  Documentation could use some clarity and the aforementioned shortcut.