hd-zero/hdzero-goggle #147: Feature request: Volume control for line-in microphone

Yith1, 2023-01-31:
My line-in microphone is saturated in the video recordings. It needs some attenuation somehow.

eg: https://www.youtube.com/watch?v=h2ZL-sFdH1s

Cascasbah, 2024-04-12:
I agree, the microphone feature can be very useful, but the lack of volume control can make it unusable at times.
Additionally, there is no way to control audio volumes during playback. Given that the goggles record audio, it can be useful to have this feature to review footage is noisy environments.

xer271, 2024-11-10:
Made some progress on this and put my info in the newer issue that is covering inputs and outputs: https://github.com/hd-zero/hdzero-goggle/issues/404#issuecomment-2466972111