hd-zero/hdzero-goggle #151: Fat32 filesystem corruption when using the switch or unplugging the goggles

AlexandreBonneau, 2023-02-03:
As per [this](https://github.com/hd-zero/hdzero-goggle/issues/84#issuecomment-1415765053) request, I'm posting again here the findings about the Fat32 formatted SD card being uncleanly unmounted, which lead to the fs to be put permanently in read-only mode.

The original messages are [here](https://github.com/hd-zero/hdzero-goggle/issues/84#issuecomment-1402771496) and [there](https://github.com/hd-zero/hdzero-goggle/issues/84#issuecomment-1403175095).

Message one:
To add more info on the bug, it seems the problem comes from the SD card being mounted read-only.
At least that's my guess is happening on the goggles, since when I insert the card in my computer (running Linux), I get the following kernel messages:
```
[65778.322421] usb 3-2: new high-speed USB device number 6 using xhci_hcd
[65778.480975] usb 3-2: New USB device found, idVendor=14cd, idProduct=1212, bcdDevice= 1.00
[65778.480982] usb 3-2: New USB device strings: Mfr=1, Product=3, SerialNumber=2
[65778.480984] usb 3-2: Product: Mass Storage Device
[65778.480986] usb 3-2: Manufacturer: Generic
[65778.480987] usb 3-2: SerialNumber: 999999999999
[65778.481583] usb-storage 3-2:1.0: USB Mass Storage device detected
[65778.481716] scsi host3: usb-storage 3-2:1.0
[65779.494927] scsi 3:0:0:0: Direct-Access     Mass     Storage Device   1.00 PQ: 0 ANSI: 0 CCS
[65779.495230] sd 3:0:0:0: Attached scsi generic sg2 type 0
[65779.724083] sd 3:0:0:0: [sdc] 487325696 512-byte logical blocks: (250 GB/232 GiB)
[65779.724173] sd 3:0:0:0: [sdc] Write Protect is off
[65779.724176] sd 3:0:0:0: [sdc] Mode Sense: 03 00 00 00
[65779.724260] sd 3:0:0:0: [sdc] No Caching mode page found
[65779.724262] sd 3:0:0:0: [sdc] Assuming drive cache: write through
[65779.725838]  sdc: sdc1 sdc2
[65779.726010] sd 3:0:0:0: [sdc] Attached SCSI removable disk
[65785.108260] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[65799.277625] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277631] FAT-fs (sdc1): Filesystem has been set read-only
[65799.277636] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277638] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277639] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277641] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277642] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277643] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277644] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277645] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[65799.277646] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
```

Looks like FAT32 does not like when you switch off or unplug your battery from the goggles, hence creating the usual `not properly unmounted` error message.

One solution perhaps could be to use a newer filesystem for the SD Card? Either exfat for windows users, or f2fs/ext4/btrfs/xfs/etc. for Linux users?

Message two:
For info, I was able to 'fix' the SD card by running fsck:
```
# fsck /dev/sdc1
fsck from util-linux 2.38.1
fsck.fat 4.2 (2021-01-31)
There are differences between boot sector and its backup.
This is mostly harmless. Differences: (offset:original/backup)
  65:01/00
1) Copy original to backup
2) Copy backup to original
3) No action
[123?q]? 1
FATs differ but appear to be intact.
1) Use first FAT
2) Use second FAT
[12?q]? 1
/isp0_0_0_0_ctx_saved.bin
  Contains a free cluster (728715). Assuming EOF.
/isp0_0_0_0_ctx_saved.bin
  File size is 40472 bytes, cluster chain length is 0 bytes.
  Truncating file to 0 bytes.
Reclaimed 2 unused clusters (65536 bytes).
Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
1) Remove dirty bit
2) No action
[12?q]? 2
Free cluster summary wrong (1382323 vs. really 1382325)
1) Correct
2) Don't correct
[12?q]? 1

*** Filesystem was changed ***
The changes have not yet been written, you can still choose to leave the
filesystem unmodified:
1) Write changes
2) Leave filesystem unchanged
[12?q]? 1
/dev/sdc1: 87 files, 714313/2096638 clusters
```

Then I was able to fly 4 batteries and get the first 2 recordings; the last 2 flights were not recorded.
Now when I plug the SD card in my computer, I again see the usual errors:
```[95250.360108] usb-storage 3-2:1.0: USB Mass Storage device detected
[95250.360234] scsi host3: usb-storage 3-2:1.0
[95251.377167] scsi 3:0:0:0: Direct-Access     Mass     Storage Device   1.00 PQ: 0 ANSI: 0 CCS
[95251.377404] sd 3:0:0:0: Attached scsi generic sg2 type 0
[95251.662897] sd 3:0:0:0: [sdc] 487325696 512-byte logical blocks: (250 GB/232 GiB)
[95251.662998] sd 3:0:0:0: [sdc] Write Protect is off
[95251.663002] sd 3:0:0:0: [sdc] Mode Sense: 03 00 00 00
[95251.663090] sd 3:0:0:0: [sdc] No Caching mode page found
[95251.663093] sd 3:0:0:0: [sdc] Assuming drive cache: write through
[95251.664859]  sdc: sdc1 sdc2
[95251.665065] sd 3:0:0:0: [sdc] Attached SCSI removable disk
[95253.976447] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[95257.202653] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202659] FAT-fs (sdc1): Filesystem has been set read-only
[95257.202664] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202666] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202667] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202668] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202669] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202670] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202671] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202672] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
[95257.202673] FAT-fs (sdc1): error, fat_get_cluster: invalid cluster chain (i_pos 525333)
```

I made sure I used the goggles switch button each time, instead of unplugging the barrel connector, while I doubt it would change anything.
Somehow, doing this with the HDO2 never lead to read-only SD card; perhaps Fatshark did run an equivalent of fsck on boot?

Last thing; I'm using the beta firmware posted today on discord `HDZERO_GOGGLE-7.68.127.7.bin`.

mard89, 2023-03-16:
> xfs please

The average user wouldn't be able to mount a XFS drive on a mac or windows PC.
Also, F2FS would be logical choice should you go for a file system that is not natively supported by 99% of computers.

RoGoR1, 2023-03-16:
> > xfs please
> 
> The average user wouldn't be able to mount a XFS drive on a mac or windows PC. Also, F2FS would be logical choice should you go for a file system that is not natively supported by 99% of computers.

I don't disagree with your premise. It would be great if they supported the use of XFS, I'm not saying make it default. 

I.E. if I formatted the card to XFS or F2FS and the goggles would be happy with that. I'm a Linux only user since the year 2000.  I use BTRFS on my workstations but XFS is fast light and rock solid. No need to ever to FS checks. Perhaps F2FS is a good choice since that is the target device for this filesystem, but I trust XFS more and don't care about every last ounce of performance. FAT just sucks. The mount command from the googgles potentially only needs the kernel module and some libraries and then can handle any of these filesystems. However, since size matters and special cases have to be made for filesystems like FAT,  it's probably not so simple. I should try formatting with XFS and see if the goggles just work. I've never tried.

AlexandreBonneau, 2023-03-20:
> It would be great if they supported the use of XFS, I'm not saying make it default.

That.

For all users alike, FAT32 is an abomination.
For windows users (I'm guessing the majority), let's make exfat the *default*.
For power users, just make the goggle recognize other filesystems (f2fs, xfs, you name it) if they already exists on the SD card. You don't even need to add the `format to [f2fs|xfs]` option menu, since those users will know how to do that on their own computer.

brandonrc, 2023-05-30:
# My Plan for Addressing the Filesystem Corruption Issue

Hello everyone :wave:,

There's been an issue with HDZero's FPV goggles involving filesystem corruption due to power loss. This is particularly troublesome for SD cards which are formatted with FAT32.

The FAT32 file system, while broadly compatible, struggles with sudden power loss. This can lead to data corruption or even render the whole filesystem unusable. ExFAT, while a more modern file system, unfortunately suffers from the same lack of power loss resilience.

To fully address this, hardware modifications would ideally be made to ensure safe shutdowns or power loss recovery. However, that's not possible since the goggles are already in the hands of users.

Techniques like atomic writing or redundancy could be considered. They would provide a safeguard against data corruption, but the filesystem itself would still be vulnerable.

## Add a function that restores SD cards when inserted (Low risk improvement)

A potential solution could be for HDZero to implement a function in the firmware that automatically checks and repairs the filesystem when an SD card is inserted. This would involve using existing disk repair tools and could transparently handle minor filesystem corruptions.

## Add NTFS Support (Solution)

After reviewing a variety of file systems, I believe NTFS could be a viable option for HDZero's use case. NTFS is a journaled filesystem, which means it maintains a log of transactions. In the event of power loss or sudden disconnection, this log can be used to recover the filesystem.

EXT4 or XFS, though robust and well-suited for Linux environments, aren't natively supported by Windows or MacOS. Given that most end users are likely to be using these systems, we want to avoid adding extra steps for them to access their data.

There's good news! An open-source library providing NTFS read and write capabilities is available. This would enable NTFS support to be incorporated into the firmware.

### Steps I am going to take
- Driver Integration: The NTFS-3G driver needs to be integrated with your firmware. This involves cross-compiling the driver for your target platform, integrating it with your build system, and testing it thoroughly.

- Testing: Extensive testing would be needed to ensure your application behaves correctly with the new file system and to evaluate the performance and stability of the NTFS-3G driver in your specific use case.

In summary, by adding NTFS support and a filesystem check/repair function, HDZero could significantly mitigate the filesystem corruption issue. I will start with the first part, add a function that restores SD cards fat32 partition then I will move to enabling NTFS on my goggles and see how DVR works with the FFPack library. I'm eager to hear your thoughts!

brandonrc, 2023-05-31:
Hit a stopping point right now. 

I need to be able to build new OS.bin's so I can test some of the code changes I made. 

I need to make sure we have a few packages installed on the OS. `dosfstools` and `ntfs-3g`. 

So options.. 

- Carl reaches out and allows me to build OS.bin files I can update my goggles for dev testing
- Root the goggles myself to install packages... or find out if someone else has a way in to where we can add binaries to the goggles. 

Any information would help me continue my work fixing the fat32 corruption issues.

RoGoR1, 2023-06-01:
I understand the new wifi module allows ssh. I am not sure as root but it does have a running ssh client and a user/password that allows management of the movies files. You could probably check the dependencies like that.  I haven't looked much into setting up the developer environment mostly because I realize it requires some special hardware that I don't have in order to connect to the goggles. I have ordered the new pcb board about a month ago but I'm not going to see it for a while.

SumolX, 2023-06-03:
Try my PR to see if it resolves the issue https://github.com/hd-zero/hdzero-goggle/pull/269

Full bin install required

christhomas, 2023-08-02:
> I don't disagree with your premise. It would be great if they supported the use of XFS, I'm not saying make it default.
>
> I.E. if I formatted the card to XFS or F2FS and the goggles would be happy [...]

nobody in their right mind would use this as a filesystem unless you're an advanced linux user, it's so niche that a significant quantity of linux users don't even know what XFS is

Exfat is the filesystem to go with, but that comes with its own problems. It can be easily corrupted and only has a single copy of the FAT therefore once it becomes corrupted, it can't be easily fixed. Fat32 is more resilient than exfat in my experience. However this issue with being unable to mount filesystems, it's an incredibly annoying issue.

AlexandreBonneau, 2023-08-02:
Well, I'll refer to [myself](https://github.com/hd-zero/hdzero-goggle/issues/151#issuecomment-1475747916); exfat is in my humble opinion the default to use, since I assume most users are using Windows.
However I fail to see why preventing user choices is still a thing in 2023 @christhomas ; if a user is comfortable using xfs, let it.
I'm pretty sure adding a few linux filesystem as a kernel module won't add too much in term of size, and as module they won't use any more memory.

RoGoR1, 2023-08-02:
Advance Linux users _are_ the only ones in their right minds ;). To be
able to mount an xfs filesystem from a Linux OS is not a far stretch. It's
not a big deal to me. As of now fat is working fine with the latest
patches. However, a robust filesystem like xfs is the icing on the cake.

I have not owned windows since I bought the only laptop I ever owned, back
in 1999 windows NT, and I replaced that with Linux. Linux == the majority
of operating systems on the Internet. So to think running Linux is radical
is silly. The mindsets to support windows users only, and on occasion Mac,
it's the foundation of the problem. The primary attraction to me of HDZero
was the Open source nature and Linux OS the goggles would be and are based
on. _Of_course_ I don't want to be using a 1980s DOS based filesystem.

Many RC hobbyist run Linux. To support XFS on googles, all that has to be
done is include the Linux module, it's light, and it's free. I'm not
asking that the goggles format function use xfs, just simply allow
mounting of the filesystem. Of course, because of the nature of FAT, and
the extra things needed to do to support it, means that there probably has
to be extra logic in the maintenance scripts, like, don't try repairing a
no DOS base filesystem. I understand that it's not completely free and
simple.

I don't much care to fight on this, that was my only two cents months ago
when the question was raised. I wasn't looking to be insulted just for
offering my opinion. Tell me I am not in my right mind. I'm not going to
argue about who is more rational. XFS is a good, robust, fast, light, no
frills, no hassle filesystem. It just takes care of itself without the
need to do filesystem checks or any maintenance at all. It doesn't get
more simple than that.

- RoGoR

On Wed, August 2, 2023 7:45 am, Christopher Thomas wrote:
> xfs please
>
>
> The average user wouldn't be able to mount a XFS drive on a mac or
> windows PC. Also, F2FS would be logical choice should you go for a
> file system that is not natively supported by 99% [...]

SumolX, 2023-08-02:
This is has been resolved by PR https://github.com/hd-zero/hdzero-goggle/pull/276

AlexandreBonneau, 2023-08-03:
> This is has been resolved by PR #276

Indeed the FAT corruption has seen a workaround (while the corruption can and will still occur since it's still FAT), which so far is working ok.

It would be better to stop talking about the mediocre default filesystem here, and add to the existing [feature request](https://github.com/hd-zero/hdzero-goggle/issues/57) about it.

RoGoR1, 2023-08-03:
> > > > xfs please
>  No consumer linux system supports xfs out of the box. So even if you are a linux user, you're still not gonna know what this is.

This has nothing to do with the goggles anymore, and I would like to drop this. However, to be corrected XFS is not at all niche. The most standard Linux filesystem used to be EXT, EXT2, EXT3, and EXT4. But now days not only does every major Linux distribution supports XFS (and has for years), but it's become the default de-facto standard. It's faster lighter and more robust than EXT4 in most every way, and it's simple. There's no reason to choose ext4 over XFS. Personally, for years, I use BTRFS in most cases. It has so many features, like ZFS. However, BTRFS is not appropriate, it's heavy, complicated, and slow. XFS is not at all exciting, it's just an extremely light, fast, stable, and simple journaling self repairing file system. Nothing prevents it from being used on windows, mac, or any other OS, it's open source and free to the public. It's not a niche and it's not controversial either. XFS is most notably good with large files, such as video files.

closed 2024-06-07 by ligenxxxx

FrancescoManfredi, 2025-04-24:
> Indeed the FAT corruption has seen a workaround (while the corruption can and will still occur since it's still FAT), which so far is working ok.
>
> It would be better to stop talking about the [...]

So this issue is still in place?
I'm currently experiencing this exact problem. After formatting the SD card I can get some recordings but in the next session I won't anymore.  
When checking dmesg I get exactly the same error logs about the volume not being properly unmounted.  
  
I might be missing something but I don't understand how PR [#276] solves this.