hd-zero/hdzero-goggle #154: Feature Request: Better DVR filename idea (like GoPro)

sabshire, 2023-02-06:
Why not do something like GoPro does for naming the files.

1.  Maintain the filename count between SD card formats, and firmware updates.
2. When files split, use a similar mechanism to GoPro so that it is easy to determine that the files were split and go together.

So, as an example, GoPro would have a filename like
GOPR3364.mp4, and if it was split, the second, third, and so one would be named GP013364.mp4, GP023364.mp4, etc

We could have something like HDZR1234.ts, and HDZ11234.ts, HDZ21234.ts, etc 

Once you get HDZR9999.ts, it could reset to HDZR0000.ts, or could continue on to 5 digit numbering, HDZR10000.ts, HDZR10001.ts, etc, and the split files would be HDZ110000.ts, and HDZ210000.ts, etc for HDZR10000.ts, and HDZ110001.ts and HDZ210001.ts for HDZR10001.ts, etc....  I am not sure what GoPro does in this scenario.

AlexandreBonneau, 2023-02-09:
This is in my opinion the wrong way to name files, since if you do this, then you break the only information that you currently have (without a powered RTC) about the order of the files.
What you propose breaks the alphanum order.

The better solution would be to implement https://github.com/hd-zero/hdzero-goggle/issues/42, and if a split is needed, just add `_1` , `_2` etc. at the end of the filename.

ligenxxxx, 2024-06-07:
Reject.

closed 2024-06-07 by ligenxxxx