hd-zero/hdzero-goggle #155: Feature Request - Delete clip option

oldrootbeer, 2023-02-07:
Would it be possible to have a delete clip option for the playback folder. Some people may have a small sd card installed and need the space in the field.
Maybe a double click of the ok button whilst the clip is selected to delete, then a further double click to confirm deletion? (A pop up or text line would be needed saying please double click to confirm, or single click to abort for example).

krizmyass, 2023-02-24:
I'm not an experienced programmer and certainly not in C, but I am enthusiastic to learn :)

RIGHT_BUTTON_PRESS (for long click) was already implemented, I just added a function to delete the selected clip.

I will improve this with the following ideas:

- add double click
- delete confirmation
- option to delete all clips at once

The files are based on the repo from yesterday with the option to delete clips by long-pressing the right button.
[HDZGOGGLE.zip](https://github.com/hd-zero/hdzero-goggle/files/10823891/HDZGOGGLE.zip)

bkleiner, 2023-02-24:
> I'm not an experienced programmer and certainly not in C, but I am enthusiastic to learn :)
>
> RIGHT_BUTTON_PRESS (for long click) was already implemented, I just added a function to delete the [...]

Thanks for trying to contribute, please try to teach yourself [how to submit a pull request](https://opensource.com/article/19/7/create-pull-request-github).
Testing _and distributing binaries_ is best taken care off once that is in place.