hd-zero/hdzero-goggle #159: Feature Request: Ability to set input voltage manually for PD Power banks (i.e. 12v, 15v ,20/20.2v)

FPVRishi, 2023-02-08:
I can see the option to manually select the number of cells is being incorporated, but what about those using a pd power bank -some input 12v,15v and even 20v or 20.2v) Thanks

bkleiner, 2023-02-09:
I am sorry, but would be the point of that? Surely the power bank is voltage regulated, it will output the same voltage until it doesn’t.

FPVRishi, 2023-02-09:
I was thinking if there has been an addition to "add ability to manually specify lipo cell count", that a similar specification of PD input voltage might be required to help with low battery alarms etc. as PD voltage would otherwise be detected as a lipo within the same voltage range, but then trigger low alarm warnings etc. when not necessary...

closed 2023-02-10 by FPVRishi