hd-zero/hdzero-goggle #161: Feature request Accurate calibration of goggle battery

fpvwannabee, 2023-02-09:
Bit of feedback based on my use of the hdzero goggles.
I've noticed that the goggle voltage is incorrectly reporting what my goggle battery is.
At the end of this morning's session they were stating 6.2 v on the osd but the battery when tested separately was 7v

robhaswell, 2023-02-09:
Is this a high-capacity li-ion battery? The goggles draw a fair amount of power, definitely enough to sag some batteries by 0.4V per cell. You should measure the voltage of the battery while it's powering the goggles.

robhaswell, 2023-08-09:
I've also noticed that my voltage needs calibrating.

pitts-mo, 2023-08-09:
Agreed, we need an easy way to calibrate the current voltage reading.

However, These goggles consume a lot of power and I think you might be surprised to see how much voltage drop is occurring during goggle operation when you test voltage at the inside of the goggle connector VS directly at the battery... The cable length and the integrity of the connectors are MUCH more likely to influence voltage drop on HDZero goggle than most other goggles.

Master92, 2023-09-04:
As #332 has been merged, I suppose you can now close this issue 🙂

Master92, 2023-11-11:
@ligenxxxx could you be so nice and close this issue?