hd-zero/hdzero-goggle #164: Feature Request - Properly Display antenna signal strength

Akunert2009, 2023-02-11:
[image]
Can the antenna signal strength be amended to display VLQ8 properly?  See attached screenshots.

Thanks!

Zuldan, 2023-02-11:
Yep 100% need some sort of calibration like you do with Rapidfire and Fusion.

FPVRishi, 2023-02-12:
Could this also be due to an SMA issue etc - some posted on facebook that the trace had lifted etc, resulting in poor signal being received and shown on one of the antennas. This was remedied by them reflowing the joint using a bit of wire to a suitable point on the board... https://www.facebook.com/groups/hdzero/permalink/572197678141531/ and https://www.facebook.com/groups/hdzero/permalink/571644371530195/

Akunert2009, 2023-02-12:
> Could this also be due to an SMA issue etc - some posted on facebook that the trace had lifted etc, resulting in poor signal being received and shown on one of the antennas. This was remedied by [...]

I asked about this issue in the HDZero discord and it was indicated that Carl himself has experienced this and said it was a calibration issue.  I do not know if it is related to lifted traces.  Perhaps someone else has additional insight.

Zuldan, 2023-02-15:
I’m pretty sure it’s a calibration issue. 

It would be great to have something like the following in the Goggles menu.

Main Menu -> Tools 

RF Calibration
Focus Chart

cynfewl, 2023-02-22:
Does this issue primarily present as lower signal strength on the left two antenna icons? I have this event at very close range and am wondering if it's this calibration issue or physical damage. I don't get zero bars, but the left 2 icons are generally about half the ones on the right.

Is there information somewhere on which antenna the icons refer to?

Zuldan, 2023-02-23:
My left two antenna icons are also usually lower than the right two but only by 1 bar. Exactly what is shown in the original post above. I remember seeing somewhere that the HDZero engineers have confirmed it's not a hardware issue, but a firmware issue. 

There are very very rare cases where the antenna adapter hasn't been soldered on correctly but I think in that case you would see 1 bar to nothing, all the time.