hd-zero/hdzero-goggle #167: goggle crash while in playback menu and press right button

adamg-86, 2023-02-19:
My goggle crash when I am in the playback menu overring over a video and I press the right button. 
FW version: 68-127-8.0.0

bkleiner, 2023-02-19:
well, crap. this somehow slipped through the cracks. 
potential fix https://github.com/hd-zero/hdzero-goggle/pull/168
with builds here https://github.com/hd-zero/hdzero-goggle/actions/runs/4217244154

adamg-86, 2023-02-19:
> well, crap. this somehow slipped through the cracks. potential fix #168 with builds here https://github.com/hd-zero/hdzero-goggle/actions/runs/4217244154

It fixed it 👌 thx for the good work.

bkleiner, 2023-02-19:
thanks. fix is merged. i will hold off for a little while, in case something else pops up, and then push a hotfix release.

closed 2023-02-19 by bkleiner