hd-zero/hdzero-goggle #180: Some people reporting latest fw is recording mp4 dvr even when ts is selected in menu.

FPVRishi, 2023-02-27:
Some people reporting latest fw is recording mp4 dvr even when ts is selected in menu in hd-zero goggle discord (2 people noted this on 26/02/23).

"daggersbaggers — Yesterday at 21:22
Has anyone had the dvr record mp4 when ts is selected? All of my dvr recordings from today are mp4 and ts files were recorded yesterday. Ts is still selected in the menu. Is this a known bug?"

"Zuldan — Yesterday at 21:24
I noticed this yesterday as well"

Zuldan, 2023-02-27:
Here is an example of the Goggle all of a sudden recording in .MP4 when it's selected as .TS in the Goggle menu. The only thing different with my setup is auto record is set to Manual and I'm using the Backpack functionality to record when the quad arms.

[image]
[image]

zhiquanyeo, 2023-02-27:
I saw something similar. I have Auto Record on, and use ELRS backpack to change channels. I powered off between every 1-2 packs and while most of the DVR files are TS (which i have it set to), I had a couple that were recorded as MP4

closed 2024-06-19 by ligenxxxx