hd-zero/hdzero-goggle #189: Stuttering at 60hz HDMI mode

RecifeStargazer, 2023-03-02:
As the title say. Cant use a walksnail vrx at 60hz. Also can't use It on my pc.

robhaswell, 2023-03-02:
Which firmware version are you using? Prior to v8.0.0 there was stuttering at 1080p60.

RecifeStargazer, 2023-03-02:
> Which firmware version are you using? Prior to v8.0.0 there was stuttering at 1080p60.

The latest 8.0.1

RecifeStargazer, 2023-03-02:
Can you confirm that stuttering was gone on 8.0.0? I can downgrade It, for sure. But the issue is happening here with 8.0.1

robhaswell, 2023-03-02:
> Can you confirm that stuttering was gone on 8.0.0? I can downgrade It, for sure. But the issue is happening here with 8.0.1

I won't be able to personally test it for a while, hopefully someone else can confirm.

appleguy010, 2023-03-05:
I AB tested against a 60hz monitor on both 8.0.0 and 8.0.1 and the issue is present on both.
Also for me when using the goggles for a prolonged period of time at 60hz I get motion sick in sim or other 3d games.

kakashi2018, 2023-03-06:
Yes, it is happening on 8.0.1 and I believe it is a known issue and I hope fix will be delivered in next release

oldrootbeer, 2023-04-11:
> Is this done by design to make the low framerate HDZ 1080p mode look better?

I doubt it. It's likely on the back burner until more pressing issues are fixed etc

RecifeStargazer, 2023-04-20:
The issue still happening with batch 1 goggles and latest firmware (13042023)

rockmv, 2023-04-24:
Unfortunately I have the same problem. On Saturday I wanted to use the goggles properly for the first time and it was terrible! After updating the Walksnail VRX and the HDZero goggles to the latest software, I got further. But only with 720p, 25mbit and 100hz. I've already ordered a new cable out of sheer desperation, but that's unlikely to be anything. Many will certainly have this problem, because the option Walksnail VRX and HDZero Goggles are a great addition if it really works. I'm about to sell the goggles again after such a long wait.

RecifeStargazer, 2023-04-24:
HDMI input working properly is a BASIC thing. It should be fixed by now!

ligenxxxx, 2023-04-25:
I don't use Walksnail VRX, but when I connect the pc to goggle hdmi in, no problem.

appleguy010, 2023-04-25:
> I don't use Walksnail VRX, but when I connect the pc to goggle hdmi in, no problem.

I know it doesn't look like a problem from first impressions, but something about the frame pacing is off. It's like the goggles are presenting different parts of the frame at inconsistent intervals.
This inconsistent frame pacing creates stuttery motion that is very distracting and nauseating when using hdmi 60hz for long periods of time.

If I'm just consuming media I don't mind it, but when gaming or using the sim it's not usable for me.

dbloemhard, 2023-04-25:
> I don't use Walksnail VRX, but when I connect the pc to goggle hdmi in, no problem.

As mentioned in the previous reply, the image shows up fine, however, if you start to fly (using the Walksnail VRX, or with a sim on your PC) and start yawing around you will see a very noticeable stutter in the image. 
I don't have one to check, but I believe the HDZero VRX also outputs 60hz right? If so, then the problem should be visible there too if they are connected to the goggles.

Other goggles will show the 60hz output from the Walksnail VRX with no stuttering, so I don't believe the problem is in the VRX.

RecifeStargazer, 2023-04-25:
> I don't use Walksnail VRX, but when I connect the pc to goggle hdmi in, no problem.

Can you kindly show me where the hdmi related files are in the goggles?

RecifeStargazer, 2023-04-25:
It seems that when detecting hdmi input, the code only distinguishes between 1080 and 720p. However i can see that at 720 its just optimized to 100fps (g_hw_stat.vdpo_tmg = HW_VDPO_720P100). However when it detects 1080p the g_hw_stat.vdpo_tmg variable is refering to a unusual frequency ( HW_VDPO_1080P50). As the code is now, is not possible to fix all issues regarding 60fps mode without a refactor but by editing g_hw_stat.vdpo_tmg = HW_VDPO_1080P50 -> g_hw_stat.vdpo_tmg = HW_VDPO_1080P60  could solve the problem? Because if this works fine on 1080p, no one would complain to use 720p100.
[image]

oldrootbeer, 2023-04-25:
> I don't have one to check, but I believe the HDZero VRX also outputs 60hz right? If so, then the problem should be visible there too if they are connected to the goggles.
>
> Other goggles will show [...]

I'll try when I'm back home in a few days. Unless someone beats me to it.

It's interesting that the vrx had timing issues with the hdo2 (random low frame rates) and it was blamed on the hdo2

RecifeStargazer, 2023-04-26:
[image]
this is the code that writes the frequency on the fpga. the sad news is that this function, I2C_Write, only accept as parameter, uint_8 format. You're basically writing the divider t on f=1/t . if t = 0x14(20), f = 50hz. A round number. However, for f=60hz, t must be 16,66666.... And its not possible to pass a floating point hex to that function. And this 0.6666 residue... is what cause the stuttering sensation. i dont know if the fpga accepts writing floating points.
@SX is there a documentation about that fpga? If it accepts a write operation with floating points, i could just create a function that does the trick, to make the precision good enough, like 60,000002hz or something like that

RecifeStargazer, 2023-04-26:
Another thing that worth to mention... the way the code deals with that is way different than when using hdz feed. The hdmi in just set this register. But the hdz modes adjusts also the oled screens frequencies and stuff.

dbloemhard, 2023-04-27:
Please ensure that 720/60 is also supported - this requires lower bitrate so ends up with a more stable image for Walksnail. Ideally i would prefer to use this as i do not need high framerate 720 if it means smearing issues.

rockmv, 2023-04-27:
I would really like support higher than 720p/60Hz, because I saw the Goggles as a multi-talent and as an alternative to DJI. And above all, the hardware should be able to use 1080p/60Hz. Even with a high frame rate.

Dragony, 2023-04-27:
I'm getting this tearing on 720p60fps HDZero native, with a Whoop VTX and the Orange Runcam camera. Is this related to this issue?

[image]

oldrootbeer, 2023-05-06:
Any progress on this?

RecifeStargazer, 2023-05-06:
> Any progress on this?

No. Tried a couple of things... But seems that the issue is that hdzero configures the OLED in a different way when using HDMI.
Trying to figure ou

RecifeStargazer, 2023-05-06:
This still happening. Should not be closed ...

gonebdg, 2023-05-07:
Hello guys,
**How to replicate this issue ?**
I've been using HDZero goggle for a month now.
Currently I only use HDZ Whoop VTX also Walksnail Avatar HD Mini kit 1s with Avatar VRX.
My HDZ goggle is running with default firmware v9.0.1and the Avatar VRX & VTX is running with v29.33.16.
So far so good, I didn't have a problem running both system with HDZ goggle (also no problem with my analog TBS Fusion).
A bit worry that I might get the same issue if I start to use both system for my 5" quad.

dbloemhard, 2023-05-07:
Set the walksnail VRX to standard framerate, and try to fly. Notice the jittering and stuttering in the image (mostly noticeable when yawing I guess)

It's not quite as bad as the analog signal disruption which can freeze and totally lose the image. But standard framerate is 60fps and should be very smooth. It's perfectly fine on other goggles.

anton-rothwell, 2023-06-01:
I can confirm this is still happening with batch 2 goggles and 9.0.1 firmware. Ideally I want 1080 and 720p 60 modes to work as these will be the modes I use for long range.  Low latency 100fps works fine but this is not my priority for walksnail use.

gonebdg, 2023-06-02:
Ah yes ... I remember I've this problem when using walksnail VRX in 1080p standard framerate and I though it was caused by the walksnail VRX firmware issue (I still use 29.x.x in my ws vrx)
Since then, I just use 720p high framerate

anton-rothwell, 2023-06-02:
I upgraded walksnail from 29 .x.x to 32.x.x today an have just test flown.  It's still a problem in 720p standard mode and is un-flyable like this. @RecifeStargazer identified the root cause of this and it's mismatched fps rates in the HDZero goggle firmware that is the problem.

Squaal, 2023-06-03:
> [image] this is the code that writes the frequency on the fpga. the sad news is that this function, I2C_Write, only accept as parameter, uint_8 format. You're basically writing the divider t on [...]

I don't think this is right. Register 0x8d is set to 0x14 on every single mode apart from HMDI in, and only when the pixel clock is low.  That function reads the IT66021 for the pixel clock frequency, this would help in decoding the image in the various modes it support on the FPGA side as well as the different color spaces. The pixels are decoded when the line on the IT66021 is pulled low, this means that there is no need to set a frequency value on the FPGA.

Rather, the modes seems to be written to 0x80, looking through it : 
- 90fps is 0x03
- 60fps is 0x01
- 50fps is 0x00
- 30fps is 0x04

On HDMI in, if the VTMG is detected as 1080p, it is assumed to be 1080p50 and it writes 0x00 (same as 720p50) to the FPGA. If it is 720p, it is assumed as 720p100 and writes 0x80, which is unique. As far as I can tell, there is simply no config for 1080p60 but I can very well be mistaken.

RecifeStargazer, 2023-06-04:
> I don't think this is right. Register 0x8d is set to 0x14 on every single mode apart from HMDI in, and only when the pixel clock is low. That function reads the IT66021 for the pixel clock [...]

I Hope you're right. That's really a clever inspection you did in the code. And easy to check up. I'll change the value written on 0x80 from 0x00 to 0x01 at line 506 and test. If It fix It, kudos for you, Bro! The next step would be improve the 720p to work with both 100 and 60 FPS.

Squaal, 2023-06-04:
Please let me know if it works, I don't have the means to test it with a Walksnail VRX as of now but I would like to see this getting fixed for everyone.

I'll try to dig a bit deeper and document the configs of the HDMI receiver/FPGA in various modes when I get some time. There is some constants about a 1080p60 mode but they are not used (HW_VDPO_1080P60)

RecifeStargazer, 2023-06-08:
> Please let me know if it works, I don't have the means to test it with a Walksnail VRX as of now but I would like to see this getting fixed for everyone.
>
> I'll try to dig a bit deeper and document [...]

I've tested the proposal solution. It not solved the problem. Instead i got a black screen.

rockmv, 2023-06-16:
I really hope that the problem will be fixed soon. While it's fine to fly at 720p, knowing that 1080p could go isn't pretty.

oldrootbeer, 2023-06-30:
> I don't use Walksnail VRX, but when I connect the pc to goggle hdmi in, no problem.

Hi Ligen.

Is there anything in the works for this issue? We have 1080 displays after all. It would be nice to use them without this problem.

RecifeStargazer, 2023-06-30:
I bet that If you pick a hdzero vrx, and use It with hdz goggles, you Will be able to see the problem.
Ligen, If i could get some documentation,  i could investigate It myself

dischinator, 2023-07-03:
> I bet that If you pick a hdzero vrx, and use It with hdz goggles, you Will be able to see the problem.

you can also hook goggles to a computer, set a 60fps resolution, you won't notice it on desktop, but if you run a sim or a game its easy to spot.

vernelv, 2023-08-06:
still having this issue at the moment. 1080p60Hz and 720p60Hz both are unusable. Frame skipping still exist while 720p100hz is the only one usable.

oldrootbeer, 2023-08-10:
Just noticed 9.2.1 rc1
#321 says that the hdmi in has been optimised. 720p 60, 720p 100 and 1080p 100.
Are there any plans for 1080p 60?
I'll update and test the 720p 60 tomorrow with ws

rockmv, 2023-08-11:
> Just noticed 9.2.1 rc1 #321 says that the hdmi in has been optimised. 720p 60, 720p 100 and 1080p 100. Are there any plans for 1080p 60? I'll update and test the 720p 60 tomorrow with ws

I'm so excited to see if this really works with Walksnail VRX and 60fps.

oldrootbeer, 2023-08-11:
I haven't flown 9.2.1 rc1 (only tested by hand flying), but it looks like 720p60, (as well as 720p100) for ws hdmi in is now smooth. Hopefully I'll test fly in a few hours.
1080p60 most definitely isn't smooth yet

dischinator, 2023-08-11:
> I haven't flown 9.2.1 rc1 (only tested by hand flying), but it looks like 720p60, (as well as 720p100) for ws hdmi in is now smooth. Hopefully I'll test fly in a few hours. 1080p60 most definitely isn't smooth yet

did you flash the pipeline build from #321 PR?

oldrootbeer, 2023-08-11:
> 

Yes. rc1 from the releases page

dischinator, 2023-08-11:
Ah I see, its already in the low band release, nice

anton-rothwell, 2023-08-28:
I've just updated goggles and can confirm, 720p is now how it should be - not flown yet, but it's noticably smoother.  1080p60 is still jittery.  This probably suits most of my use cases, but is still not great for gaming.

appleguy010, 2023-09-02:
Just tried the new cd23ede test firmware. It is supposed to fix this exact issue, but from my testing the 1080p60 mode looks exactly like the 720p60 mode. There is little detail and the image is very blurry. It is very clear that it is not the true resolution of the panel.
Switching to 1080p50 and the image is super sharp.

I hope HDzero doesn't see the issue as "fixed" after this update because it clearly isn't.

cougman, 2023-09-02:
Why doesn’t the in line mic work with the analog mode?

Sent from myMail for iOS

oldrootbeer, 2023-09-02:
> Just tried the new [cd23ede](https://github.com/hd-zero/hdzero-goggle/commit/cd23ede2333cb4617f4c37f6a1ac112df2a4dede) test firmware. It is supposed to fix this exact issue, but from my testing the [...]

Are you sure? Is this the fw that was posted on Discord a week or so ago? It looks pretty good to me when I tested with a movie.

appleguy010, 2023-09-02:
> 
> Are you sure? Is this the fw that was posted on Discord a week or so ago? It looks pretty good to me when I tested with a movie.

Yea that's the firmware I'm using. I will try flashing the firmware again in case there is something borked with my install.

Will you try 1080p50 and 1080p60 back to back? Curious to know if your goggles behave like mine.

oldrootbeer, 2023-09-02:
> Yea that's the firmware I'm using. I will try flashing the firmware again in case there is something borked with my install.
>
> Will you try 1080p50 and 1080p60 back to back? Curious to know if your [...]

Yes I will 👍

dischinator, 2023-09-03:
Maybe using a test image like PM5644 https://commons.wikimedia.org/wiki/File:PM5644.svg could help. You should notice a difference between real 1080p and lower resolutions.

dischinator, 2023-09-03:
> Maybe using a test image like PM5644 https://commons.wikimedia.org/wiki/File:PM5644.svg could help. You should notice a difference between real 1080p and lower resolutions.

Interesting, I just tried the PM5644 image on cd23ede. It looks fine on 1080p60 but on 1080p50 the smallest set of horizontal lines are one gray block, so the 50hz mode seems scaled.
Just for reference I also checked 720p and it looks like 1080p50.

1080p60:
[image]

1080p50:
[image]

appleguy010, 2023-09-03:
> Interesting, I just tried the PM5644 image on [cd23ede](https://github.com/hd-zero/hdzero-goggle/commit/cd23ede2333cb4617f4c37f6a1ac112df2a4dede). It looks fine on 1080p60 but on 1080p50 the [...]

 Very interesting, it sounds like the reverse of my experience.
The eizo sharpness test is also a good one for this. ClearType is off so it makes the difference between the two easy to spot.
https://www.eizo.be/monitor-test/

dischinator, 2023-09-04:
> Very interesting, it sounds like the reverse of my experience.
> The eizo sharpness test is also a good one for this. ClearType is off so it makes the difference between the two easy to spot.

Ah yes the monitor test is also a good idea. Could you confirm with the monitor test your goggle has the scaled res for 1080p60? That would be really strange. Also 50hz and 60hz should be same anyways, either scaled or native.

RecifeStargazer, 2023-12-09:
This fixes the issue: 
https://github.com/hd-zero/hdzero-goggle/pull/380
I'll close It now

RecifeStargazer, 2023-12-09:
https://github.com/hd-zero/hdzero-goggle/pull/380

closed 2023-12-09 by RecifeStargazer